FastLED/FastLED #934: Why is there a delay() at the start of setup()?

SteveRMann, 2019-12-08:
Every sample file and every demo sketch I have looked at has a delay(1000) [sometimes 2,3 or 4-seconds] at the start of the setup function.

Why?

Why is a delay required at the start of setup()?  Or is it?

SteveRMann, 2019-12-09:
No one?  Is it done simply because "that's the way we've always done it"?

kriegsman, 2019-12-09:
It's there because in some situations, if your code suddenly lights up a lot of LEDs very brightly, it causes a voltage drop across the microcontroller, and the MCU goes into a "brownout reset" loop.  If you're in the lab, and can easily disconnect the LEDs and reprogram the MCU, you don't need any delay -- just to unplug the LEDs.  But if you _can't_ easily unplug the LEDs from the MCU, e.g because it's all soldered together, or inside some enclosure, or high up on a wall, or out in the desert, having a couple of seconds to quickly start the reflash process can be enough to save the day.  Dan and I both had the bad situation happen numerous times, and so we just started putting a startup delay in the beginning of all of our sketches.

It's not _needed_.  It's insurance.  And how much insurance _you_ need depends on your circumstances.  Use your own judgement.  But as a general matter, it's not a bad place to start out with a few second delay before lighting any LEDs, for reasons described above.

Alternative: have a slow ramp up in FastLED.setBrightness(...) over the first few seconds.  That's what I sometimes do now with wearables and other things where I want to see _some_ light right away when I turn it on, but also want the insurance of being able to escape the brownout reset loop.

Thanks for asking about this.  For general questions like these, please check out the FastLED discussion on Reddit: https://www.reddit.com/r/FastLED/

closed 2019-12-09 by kriegsman

SteveRMann, 2019-12-10:
Thanks.  So if my LEDS are on a separate PSU that is more than sufficient to drive all LEDS white at full brightness, then the delay is not needed.

kriegsman, 2019-12-10:
I’d say: Yes, it’s not needed in that case.