FastLED/FastLED #938: Unable to complile for Adafruit nRF52840 feather

ohiodev007, 2019-12-13:
I am new to the arduino space and am hopeful that I am making a rookie mistake here but I've been all across the web and cannot for the life of me find a resolution to my issue. This is leading me to believe there is an issue.

I am working on a project that makes use of the Adafruit nRF52840 feather and WS2812B LEDs. I am using the Arduino IDE version 1.8.10, Adafreuit nRF52 board management library 0.14.6 and the latest release of FastLED 3.3.2.

I would like to note I've tested this setup using the Adafruit NeoPixels library and everything runs great.

To test FastLED out I am trying the Cyclon.ino file from the examples directory verbatim (https://github.com/FastLED/FastLED/tree/3.3.2/examples/Cylon).

Strangely enough I am receiving the following compiler errors.

`In file included from C:\Code\test.ino:14:0:

C:\Users\...\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.002

 #    pragma message "FastLED version 3.003.002"

                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52.h:324:0,

                 from C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastled_arm_nrf52.h:6,

                 from C:\Users\...\Documents\Arduino\libraries\FastLED/platforms.h:11,

                 from C:\Users\...\Documents\Arduino\libraries\FastLED/FastLED.h:52,

                 from C:\Code\Mobibooth\LED Controller\mobibooth_cruise_controller\mobibooth_cruise_controller.ino:14:

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:62:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 0, 25, 0); // D0  is P0.25 -- UART TX

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:64:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 2, 10, 0); // D2  is P0.10 -- NFC2

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:65:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 3, 47, 1); // D3  is P1.15 -- PIN_LED1 (red)

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:66:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 4, 42, 1); // D4  is P1.10 -- PIN_LED2 (blue)

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:67:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 5, 40, 1); // D5  is P1.08 -- SPI/SS

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:68:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 6,  7, 0); // D6  is P0.07

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:69:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 7, 34, 1); // D7  is P1.02 -- PIN_DFU (Button)

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:72:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 8, 16, 0); // D8  is P0.16 -- PIN_NEOPIXEL

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:73:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN( 9, 26, 0); // D9  is P0.26

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:74:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(10, 27, 0); // D10 is P0.27

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:75:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(11,  6, 0); // D11 is P0.06

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:76:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(12,  8, 0); // D12 is P0.08

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:77:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(13, 41, 1); // D13 is P1.09

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:78:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(14,  4, 0); // D14 is P0.04 -- A0

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:79:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(15,  5, 0); // D15 is P0.05 -- A1

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:82:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(16, 30, 0); // D16 is P0.30 -- A2

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:83:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(17, 28, 0); // D17 is P0.28 -- A3

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:84:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(18,  2, 0); // D18 is P0.02 -- A4

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:85:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(19,  3, 0); // D19 is P0.03 -- A5

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:88:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(22, 12, 0); // D22 is P0.12 -- SDA

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:89:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(23, 11, 0); // D23 is P0.11 -- SCL

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:92:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(24, 15, 0); // D24 is P0.15 -- PIN_SPI_MISO

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:93:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(25, 13, 0); // D25 is P0.13 -- PIN_SPI_MOSI

               ^

C:\Users\...\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastpin_arm_nrf52_variants.h:94:15: error: expected constructor, destructor, or type conversion before '(' token

     _FL_DEFPIN(26, 14, 0); // D26 is P0.14 -- PIN_SPI_SCK

               ^

Multiple libraries were found for "FastLED.h"
 Used: C:\Users\...\Documents\Arduino\libraries\FastLED
exit status 1
Error compiling for board Adafruit Bluefruit Feather nRF52840 Express.
`

This is leading me to believe that there must be a definition missing somewhere. I am hoping that this is a simple fix and the issue can be remedied quickly.

slootsky, 2019-12-13:
If I got that error, I'd making sure I wasn't missing a semicolon somewhere before the #include of the fastled library.

But if you took the example verbatim, I'm not sure.  Although, the example has the #include on line 1, and your error message indicates the #include is on line 14.

ohiodev007, 2019-12-16:
The error is in the library file and not the code referencing it. As noted I am using the sample files provided with the library.

baramuse, 2019-12-19:
may be same issue than for the NRF52832 (#919)
For me that : https://github.com/FastLED/FastLED/issues/919#issuecomment-553543763  fixed it :

> Uncommenting the #define entries in fastpin_arm_nrf52.h :
> #define NRF_P0_BASE   0x50000000UL
> #define NRF_P1_BASE   0x50000300UL

henrygab, 2019-12-20:
This has a correct fix in a PR that's reviewed and ready: #900 

The problem is straightforward, just waiting for the maintainers to review / approve the PR.

ohiodev007, 2019-12-23:
@henrygab thanks!

gregsqueeb, 2020-01-03:
@henrygab I am having this same issue and have tested that PR on this board and it is still giving me the same errors. Any other ideas?

gregsqueeb, 2020-01-03:
I got it to compile. I think someone might have forgotten an underscore.

I changed this line in fastpin_arm_nrf52.h
`#define FL_DEFPIN(ARDUINO_PIN, BOARD_PIN, BOARD_PORT)    \`

to include an underscore before FL_DEFPIN, so the line now looks like this and compiles: 
`#define _FL_DEFPIN(ARDUINO_PIN, BOARD_PIN, BOARD_PORT)    \`

gregsqueeb, 2020-01-03:
@ohiodev007 This should fix things not compiling: https://github.com/FastLED/FastLED/pull/946

henrygab, 2020-01-03:
@gregsqueeb -- Well done!  This looks like a good fix to me, and it matches the use of that macro in from the other boards in this library.  I am considering opening a branch in my fork, just to combine all the nRF52-related changes, to make it easier for everyone else to find the necessary changes.  will update if/when I do that.

henrygab, 2020-01-31:
@ohiodev007  -- last fix was merged yesterday, so I think if you build directly from master now, it should compile.  (Next release should have the fixes also.)  Yay!

henrygab, 2020-02-04:
@ohiodev007 -- This should be fixed in release 3.3.3.  Can you confirm / close if this resolves the issue for you?  Thanks!

kriegsman, 2020-02-17:
I'm going to close this, as I think it is working now.  If there are further problems we can reopen it, or start a new ticket for the new issues.
Thanks for the report, and the help getting it working.

closed 2020-02-17 by kriegsman