FastLED/FastLED #939: Wiki missing much documentation

altimmons, 2019-12-15:
Im working on learning the ins and outs of FastLED.  I usually always take notes.  Id like to flesh out the wiki where needed, but it appears locked on the project.  If you are interested in it being expanded upon with some critical missing documentation, let me know how I can contribute,

senorblasto, 2019-12-30:
I'd be happy to help with the documentation. Been using FastLED for years have a fairly good idea of how things work

senorblasto, 2019-12-30:
I've never been let down by documentation over at fastled.io
Do you plan on migrating that to github?

I also find using code completion via VS_code as being extremely helpful.

Maybe I could create a wiki page explain how to get that set up

kriegsman, 2019-12-30:
Hi @senorblasto and @altimmons - Thank you very much for the offers to help.  As you know, we lost Dan Garcia @focalintent this past September, and I've been dealing with personal matters since then.  
However, going into the new year, we're getting things going again, and this help will be very much appreciated.  I'm leaving this ticket open and assigned to me for now as a reminder to connect after the new year.  Thanks again- -Mark