FastLED/FastLED #940: Add support for WS2815

Teddyz, 2019-12-16:
I would like you to add support for WS2815.
Using WS2813 (closest relative I found) is that sometimes the LEDs show too bright colours. This happens because the timing of a "0" is much shorter for this chipset, resulting in the LEDs sometimes mistakenly decodes a 0 as a 1. 

For WS2813, using ESP8266, FastLED outputs a 0 as a roughly 390ns pulse. WS2815 wants between 220 and 380. So most often it works, but using longer cables, different voltages, temperatures, batches/manufacturers it sometimes goes wrong.

Reference:
Oscilloscope picture of ESP8266 using FastLED sending WS2813 data
[image]

WS2815 manual
www.normandled.com/upload/201808/WS2815%20LED%20Datasheet.pdf
[image]

T´hank you!

Teddyz, 2019-12-16:
Sorry, I found that WS2812 had a shorter "0" time of 320ms. My short test showed no problems, so I suppose it will work nicely.

However, I suggest that you still add WS2815 to the list to avoid future confusion (and to make the compatible-list even more impressive).

Just for the fun of it, is there a way to play with these timings using FastLED?

OK to close.

Adding screenshot for WS2812 (or was it WS2812B)
[image]

Teddyz, 2019-12-18:
I found another problem using WS2815 with ESP8266. Running DemoReel100 works well up to 65 LEDs, if I set 66 LEDs the delay between two frames becomes too short so they are stacked after each other, instead of on top (replacing). This can be mitigated by adding an extra delay, like:
```CPP
  // send the 'leds' array out to the actual LED strip
  FastLED.show();  
  // insert a delay to keep the framerate modest
  FastLED.delay(1000/FRAMES_PER_SECOND); 
  delay(1);
```
But I would prefer that it was in the library code. :)

jandechent, 2020-01-12:
For me, it is not working at all. The below example should turn one led at a time blue, but I get red, white, green. And also the locations don't match. My led stripe is in total 120 long, but for simplicity I reduced `NUM_LEDS` to `3`. I tried `WS2812B` as well as `WS2813`. My hardware is Arduinio Nano w/ ATmega328. 

```
#include <FastLED.h>
#define NUM_LEDS 3

#define LED_TYPE WS2812B
#define COLOR_ORDER GRB
#define DATA_PIN 3
CRGB leds[NUM_LEDS];
void setup() { 
  Serial.begin(9600);
  FastLED.addLeds<LED_TYPE, DATA_PIN, COLOR_ORDER>(leds, NUM_LEDS);
}

void loop() {
  for(int dot = 0; dot < NUM_LEDS; dot++) { 
    Serial.println(dot);
    leds[dot] = CRGB::Blue;
    FastLED.show();
    leds[dot] = CRGB::Black;
    delay(1000);
  }
}
```
Anyone solved the issue yet or can point me to the right file, where the timings are defined?

Teddyz, 2020-01-12:
I tested your code on Arduino Uno + WS2815 and it works as I think is intended, a moving blue LED. Unfortunately I have no answer to you question. 

Check so you do not use BI as input, DI, the pin next to +12V is the input, BI shall be grounded. You can do this test running WS2815 on 5V. It works as long as you do not mix colours.

jandechent, 2020-01-13:
@Teddyz Yes - ground. I forgot to connect the 12V-gound to the board ground. That was the issue, obvious if I think about it. All 120 LED working perfectly for me now. 

I also suggest adding the WS2815B to the list of supported strips.

jammerxd, 2020-12-11:
I am having issues getting the data to send consistently when I need to extend the cabling to the strip from the arduino. I am using an arduino with WS2815. I'm also using an extension for the 4 pins of the LEDs from the Arduino as the LED strip is mounted on the ceiling and the arduino is housed on the floor with a power supply. Without the extension, the timing for WS2812B works fine. But with the extension, I have to call FastLED.delay(1000/15); and run the strip at 15fps compared to no limit directly attached. 

The LED strip behaves beautifully directly attached but there are so many timing issues when it's not. 

Please look into adding specific timing for the WS2815 strips. 

For reference I'm controlling a strip of 300 LEDs.

Here's my arduino source code(it reads 3 PWM signals and does a shift to push the color down the strip): https://pastebin.com/0MVtuTKv

I've even tried using an SP901E amplifier between the extension cord and the arduino.

Teddyz, 2020-12-11:
Long cables between Arduino and led-strip gives you two problems. 
First that the voltage level for the ground is not the same at both ends of the cable. You can test this by keeping the long cable and just use a few diodes of the strip instead of lighting up all of them. Low current consumption makes it work better. 

Second is that long unbalanced data cables makes the high frequency data 'fuzzy' and hard to read for rhe first led in the strip. Try putting a ~100 ohm resistor on the data lead close to the Arduino. Might help you to have fewer glitches.

Best is to have the Arduino close to the leds. I have successfully used long cables but with electronics that converts the signal to be balanced (twisted pair). Much work but it can be done. And you still need a box close to the leds.

jammerxd, 2020-12-11:
Could I use the extension the place the amplifier next to the strip and have that work? How do professional lighting crews get these long runs to work?(i.e. lighting a rooftop or lighting an arena ceiling)

Teddyz, 2020-12-12:
You can build an extension as I described above. It will work. I tested CAN transcievers with good result.

Professionals would not touch these crappy excuses for led drivers. There is professional stuff, having two-way communication with error-detection and individual colour-correction. And more. I have no deep knowledge how they do and what they use. Sorry.

marmilicious, 2020-12-12:
@jammerxd  If you have a long data run I would suggest looking into [RS-485 modules](https://www.amazon.com/s?k=max485+module).  Get a pair and put one on each end (one sending, one receiving) of some twisted pair wiring for your long run.  

https://youtu.be/Oq019vxRj54

qistoph, 2020-12-23:
My WS2815 had issues too. Most notably the pixels are all shift by 1. The first color sent doesn't show and the last LED doesn't light up. This could easily be solved by adding an extra output LED, but just felt wrong.

For my strip this seems to work quite well on a Wemos D1 mini:
```cpp
// WS2815 - 220ns, 360ns, 220ns
template <uint8_t DATA_PIN, EOrder RGB_ORDER = RGB>
class WS2815Controller : public ClocklessController<DATA_PIN, C_NS(220), C_NS(360), C_NS(220), RGB_ORDER> {};

template<uint8_t DATA_PIN, EOrder RGB_ORDER> class WS2815 : public WS2815Controller<DATA_PIN, RGB_ORDER> {};

FastLED.addLeds<WS2815, RX, GRB>(leds, NUM_LEDS);
```

steeley, 2021-02-17:
Proper support for WS2815 12v needed.

I'm using Teensy 3.2 and 3.5 and WS2815 12v(FastLED set to use WS2812) not all that reliable....
Seems when the strip is cold some LEDs  don't respond until warm. 
Many of the demo projects don't work correctly.

I have designed and built large systems with APA102C that have all worked perfectly, but these APA are not around so much
and 12v is easier to use.

Changchung, 2021-02-17:
> Proper support for WS2815 12v needed.
> 
> I'm using Teensy 3.2 and 3.5 and WS2815 12v(FastLED set to use WS2812) not all that reliable....
> Seems when the strip is cold some LEDs don't respond until warm.
> Many of the demo projects don't work correctly.
> 
> I have designed and built large systems with APA102C that have all worked perfectly, but these APA are not around so much
> and 12v is easier to use.
> 
> NOTE: WS2815 from BTF lighting seem more stable.

I am wrong or voltage have nothing to do with the library and software? Could be a data problem. Weak data.

Changchung, 2021-02-17:
I hope this help. Is a little trick to boost the data. You cut a Led and leave it near to the board.

https://hackaday.com/2017/01/20/cheating-at-5v-ws2812-control-to-use-a-3-3v-data-line/

kovalensky, 2023-10-10:
Will there be support for ws2815 in the near future?

redPanther, 2024-04-18:
Just bumping up this topic ... Would be nice for offical WS2815 support!

adictoetico, 2024-04-19:
+1 on the WS2815 support....

zackees, 2024-04-19:
This one isn't hard to implement. There are templates that take in timing parameters for the data line.

I don't have WS2815 so I can't test.

If someone does then just create a new class with the proper template params, add a compile test then tag me in your commit and i'll review.

Granjow, 2024-08-30:
@zackees Could I just get you a WS2815 strip for implementing support? :)

zackees, 2024-08-30:
Sure. I can adjust the timings for the WS2815. Can you bribe me with a large roll? I'd like to use this in a few projects.

Granjow, 2024-09-02:
Definitely! Do you want to email me your address so I can get some sent to
you? :)

zackees, 2024-09-03:
Dm me on reddit:

https://www.reddit.com/user/ZachVorhies/

zackees, 2024-09-13:
I wrote a python script for generating timings from the data sheet and now I can add all the clockless controllers.

WS2815 is now supported.

closed 2024-09-13 by zackees

redPanther, 2024-10-01:
I saw ws2815 is supported in version 3.7.7. Would be nice if documentation (https://github.com/FastLED/FastLED/blob/master/README.md and https://github.com/FastLED/FastLED/wiki/Chipset-reference) is updated in regards of new supported led chip.

thx!! for the great work!

zackees, 2024-10-03:
I've got my hands full to the next release.

Feel free to issue a PR