FastLED/FastLED #948: Tracking NRF52 pending PRs

henrygab, 2020-01-04:
Until the depot resumes taking PRs, I'm integrating those needed for the nrf52 series of chips into [my nrf52 fork](https://github.com/henrygab/FastLED/tree/nrf52_fixes), as a way to help simplify getting people up and running during this transitional time.

Currently integrated PRs:
* [x] #946 Missing underscores in `_FL_DEFPIN` causing compilation fail (regression from #866) (Issue #938)
* [x] #900 Fix compilation fail for NRF52832 (and other single-port NRF52 chips) (Issue #919)

Feel free to reference additional NRF52 related PRs within this issue, that you'd like to see integrated in the interim.

lmh91, 2020-01-07:
I think #917 should be added here?

I also have a Arduino Nano 33 BLE (nRF52840) and can't compile FastLED with the same error as in this issue.

henrygab, 2020-01-08:
#917 is a different issue … that's for support of a brand new board.  I'll follow-up on #917.

closed 2020-01-31 by henrygab