FastLED/FastLED #952: Enhancement - examples/multiple adding a mixed length example

layzsmurf, 2020-01-10:
Hello,

you already ship 6 examples to use multiple LED strips with FastLED. However all the examples use strips of the same LED length. Being totally unfamiliar with FastLED this led me to the false conclusion that FastLED has a limitation, that I need to have strips of the same length.

My suggestion would be to add one example where Length of the strips are different.

E.G.: https://pastebin.com/3UjzA66a (just an alteration of your examples)
Has an 12 LED Ring and a 5 LED Strip

marmilicious, 2020-01-13:
I agree, I like the idea of one or more examples showing you can have different lengths.  A comment in the examples mentioning different length strips can be used could also be good.

@layzsmurf  Curious, in your example, line 25, did you multiply the two length variables together so that loop runs until they become synced up again?

layzsmurf, 2020-01-20:
@marmilicious Yes, just by multiplying you get an even multiple for both - so each strip will loop without skipping an LED.