FastLED/FastLED #954: ESP8266 Crashes when controlling 1200 WS2812B

timfisc, 2020-01-19:
Hello.
Every time when I try to set all WS2812B to the color white, the ESP8266 crashes (error code 2).
I tried to fix it with Allow Interrupts and some other fixes. Still not working... Any help is appreciated!
Here's the code:

```
#define FL_ALIGN_PROGMEM __attribute__ ((aligned (4)))
#ifdef ESP8266
/*
  Spending too much time in the ISR or disabling the interrupts for too long
  time could cause WDT reset.
  For info see:
  https://github.com/espressif/ESP8266_NONOS_SDK/issues/90
  and
  https://github.com/FastLED/FastLED/wiki/Interrupt-problems
*/
//#define FASTLED_ALLOW_INTERRUPTS 0
#endif
#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <FastLED.h>
#ifndef STASSID
#define STASSID "router"
#define STAPSK  "bytimfischbach"
#endif
#define NUM_STRIPS 2
#define NUM_LEDS_PER_STRIP 900
#define NUM_LEDS NUM_LEDS_PER_STRIP * NUM_STRIPS
const char* ssid = STASSID;
const char* password = STAPSK;
bool ota = false;
const char v = '1';
const char v2 = '0';
bool ltest = true;
CRGB leds[NUM_STRIPS * NUM_LEDS_PER_STRIP];
void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() == WL_CONNECTED) {
    ArduinoOTA.setHostname("myesp8266");
    ArduinoOTA.setPassword("admin");
    ArduinoOTA.onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH) {
        type = "sketch";
      } else { // U_FS
        type = "filesystem";
      }
      // NOTE: if updating FS this would be the place to unmount FS using FS.end()
      Serial.println("Start updating " + type);
    });
    ArduinoOTA.onEnd([]() {
      Serial.println("\nEnd");
    });
    ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    });
    ArduinoOTA.onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) {
        Serial.println("Auth Failed");
      } else if (error == OTA_BEGIN_ERROR) {
        Serial.println("Begin Failed");
      } else if (error == OTA_CONNECT_ERROR) {
        Serial.println("Connect Failed");
      } else if (error == OTA_RECEIVE_ERROR) {
        Serial.println("Receive Failed");
      } else if (error == OTA_END_ERROR) {
        Serial.println("End Failed");
      }
    });
    ArduinoOTA.begin();
    Serial.println("Ready");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
    ota = true;
  }
  else {
    // put your setup code here, to run once:
    // tell FastLED there's 900 NEOPIXEL leds on pin D3, starting at index 0 in the led array
    FastLED.addLeds<NEOPIXEL, 0>(leds, 0, NUM_LEDS_PER_STRIP);

    // tell FastLED there's 900 NEOPIXEL leds on pin D4 , starting at index 900 in the led array
    FastLED.addLeds<NEOPIXEL, 2>(leds, NUM_LEDS_PER_STRIP, NUM_LEDS_PER_STRIP);
  }
}
void loop() {
  if (ota == true) {
    ArduinoOTA.handle();
  }
  //maincode
  else {
   if (Serial.available()) {
      char w = Serial.read();
      if (w == 'd') {}
      else if (w == 's') {
        Serial.print('o');
      }
      else if (w == 'v') {

        Serial.print(v);
        delay(10);
        Serial.print(v2);
      }
      else if (w == 'p') {
        Serial.print('o');
        ESP.restart();
      }
      else if (w == 'l') {
        if (ltest == true) {
          ltest = false;
          for (int i = 0; i < NUM_LEDS; i++) {
            leds[i] = CRGB(255, 255, 255);
            FastLED.show();
          }
          delay(50);
        }
      }
      else if (w == 'b') {
        for (int i = 0; i < NUM_LEDS; i++) {
          leds[i] = CRGB(0, 0, 0);
          FastLED.show();
          ltest = true;
        }
        delay(50);
      }
    }
  }
}

timfisc, 2020-01-19:
Fixed the issue. It was a coding mistake. The FastLED.show(); command should be after the for loop. It took so long to refresh every led by once that the watchdog of the esp8266 thought that the code froze.

closed 2020-01-19 by timfisc

marcmerlin, 2020-03-10:
With that many pixels, you may want to look into parallel output (4 way on ESP8266), and if you can upgrade ESP32 will give you 24 way parallel output, or more with I2S

Kamal-Sonani, 2020-05-21:
@marcmerlin by uaing 4way on esp8266 and basic calculation of ram usage what is the maximum number led we can drive & slowest frame rate (min 20 FPS).

In esp32 im using 74hc595 and I'm driving nearly 19000+ led @ 45 fps & its working fine.

marcmerlin, 2020-05-21:
Why are you asking me to do the math for you? :)
You can look up how much RAM is usable on ESP8266 and divide by 3 bytes per pixel :)
frame rate per pin will be the same on ESP8266 or ESP32