FastLED/FastLED #960: ATTINY code will never run.

8633brown, 2020-01-29:
This section of code will never run 
https://github.com/FastLED/FastLED/blob/42441aaa2448c02480846ba6015180347ac61257/lib8tion/scale8.h#L28-L60
as ```SCALE8_AVRASM``` can only equal 1 when ```LIB8_ATTINY``` is undefined.
https://github.com/FastLED/FastLED/blob/42441aaa2448c02480846ba6015180347ac61257/lib8tion.h#L245-L253

kriegsman, 2020-01-29:
Oops. Good catch. Thank you.

zackees, 2026-06-03:
**fbuild build-system status update** (tracker: FastLED/fbuild#389)

This is a FastLED-side macro-guard issue, not a build-tool issue. fbuild has no fbuild-side gap to track here. The bug is the logical inconsistency between `LIB8_ATTINY` and `SCALE8_AVRASM` in `lib8tion.h` / `lib8tion/scale8.h`.

Cross-linked from FastLED/fbuild#389 for completeness.