FastLED/FastLED #964: Adafruit Feather nRF52840 Express only controlling 144 LEDs

sandman15951, 2020-02-08:
I have having an issue where a nRF52840 Express can only control the first 144 LEDs of a 2812B strip.  This issue happens regardless of the sketch I run and regardless of what I set the number of LEDs to.  If I set the number of LEDs above 220, the LEDs stop responding all together.  In between 144 and 220, the code/strip behave as if those lights were working properly, but they don't light up.

If you want an example of a sketch I have this problem with, the Cyclon.ino file from the examples directory (https://github.com/FastLED/FastLED/tree/3.3.2/examples/Cylon) is one (changing the number of LEDs obviously).

I've tried 2 boards, they both have this problem.
It's not a power problem, even if I set the brightness way way down I still have this problem.  I have other boards that can run the same sketches at higher power levels.
Its not a problem with the rest of the hardware, other boards run just fine with the same hardware and sketches.

Anyone had issue like this before?  It's pretty strange

drewandre, 2020-02-09:
Looks like this is hard coded for nrf52 platforms: https://github.com/FastLED/FastLED/blob/master/platforms/arm/nrf52/clockless_arm_nrf52.h#L10

sandman15951, 2020-02-11:
Haha, that was easy.  Thank you!

closed 2020-02-11 by sandman15951