FastLED/FastLED #966: FastLED 3.3 compilation error on Linux

Coulomb-f, 2020-02-11:
I have downloaded FastLED 3.3 from source, given it a legal name and installed it in /home/sketchbook on my Linux Debian Buster Machine. Opening the example DemoReel100, I get tons of errors:

```
In file included from DemoReel100.ino:1:0:
/home/phil/sketchbook/libraries/FastLED3_3_3/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
                     ^
In file included from /home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/fastled_avr.h:6:0,
                 from /home/phil/sketchbook/libraries/FastLED3_3_3/platforms.h:39,
                 from /home/phil/sketchbook/libraries/FastLED3_3_3/FastLED.h:52,
                 from DemoReel100.ino:1:
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h: In static member function ‘static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showRGBInternal(PixelController<RGB_ORDER>&)’:
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:409:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:410:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:411:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:425:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:426:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:427:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:444:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:445:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before ‘__attribute__’
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/phil/sketchbook/libraries/FastLED3_3_3/platforms/avr/clockless_trinket.h:446:55: note: in expansion of macro ‘FL_FALLTHROUGH’
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
```
Is this an issue on my end somehow? I didn't find an issue similar to mine. However, compiling other examples from other libraries, including NeoPixel Strandtest.

kriegsman, 2020-02-11:
Hi and thanks very much for this report. What version of avr-gcc are you using to compile the sketch? I think we have a problem with “older” versions of gcc; knowing your version will help us figure this out quickly.

Coulomb-f, 2020-02-11:
`gcc --version` 
`gcc (Debian 8.3.0.6) 8.3.0`

I will update to 9.2 as it is the current stable version, according to Wikipedia. Will report back.

kriegsman, 2020-02-11:
Thanks for replying.  The "regular" installed version of gcc is not the one used to compile Arduino sketches.  Instead, a special version usually called "avr-gcc" is used, and it is probably not in your shell's regular search path.  It's usually installed underneath the Arduino IDE tree somewhere.

Anyway, the version number that would be helpful is the version number of whatever version of avr-gcc is installed on your system.  Also, do you know the version number of Arduino IDE that you are using?  That might help also.

Coulomb-f, 2020-02-11:
Of course, more than happy to help. So I did some research on how to find out what version of what compiler is used, here is the output: 

```
phil@debianlenovo:~$ avr-gcc -v
Using built-in specs.
Reading specs from /usr/lib/gcc/avr/5.4.0/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/avr/5.4.0/lto-wrapper
Target: avr
Configured with: ../gcc/configure -v --enable-languages=c,c++ --prefix=/usr/lib --infodir=/usr/share/info --mandir=/usr/share/man --bindir=/usr/bin --libexecdir=/usr/lib --libdir=/usr/lib --enable-shared --with-system-zlib --enable-long-long --enable-nls --without-included-gettext --disable-libssp --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=avr CFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat ' CPPFLAGS='-Wdate-time -D_FORTIFY_SOURCE=2' CXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat ' FCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong' FFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong' GCJFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong' LDFLAGS=-Wl,-z,relro OBJCFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat ' OBJCXXFLAGS='-g -O2 -fdebug-prefix-map=/build/gcc-avr-5.4.0+Atmel3.6.1=. -fstack-protector-strong -Wformat '
Thread model: single
gcc version 5.4.0 (GCC) 
```

This was a bit confusing to me, some said that different boards even use different compiler-versions. But the files pertaining to avr-gcc in the arduino tree are all symlinks.

And I'm using the arduino version from the stable branch of Buster-stable (but other repositories seem to have the same version, and usually on debian I'm not going to try experimental repos.)
https://packages.debian.org/buster/arduino

so it is version **arduino (2:1.0.5+dfsg2-4.1)**
also see https://packages.debian.org/buster/arduino

kriegsman, 2020-02-11:
Ah thank you.

You might want to pull the latest code from git again. Just merged a change that might help.

Coulomb-f, 2020-02-12:
Thank you very much. Now after pulling the latest commit, one issue resolved, another one emerged :)

The error message pertaining to clockless_trinket.h went away. However, there is another error massage preventing me from successfully compiling my sketch:

There is a function in DemoReel100 `rainbowWithGlitter` which uses fract8 as an argument.

**The code:**

```
//***********************
//### WS2912B Functions #
//***********************
void Glitter( fract8 chanceOfGlitter)
  {
    fill_rainbow( leds, NUM_LEDS, gHue, 8);
    EVERY_N_MILLISECONDS( 20 ) { gHue++; }
    fadeToBlackBy( leds, NUM_LEDS, 10);
    if( random8() < chanceOfGlitter) {
    leds[ random16(NUM_LEDS) ] += CRGB::White;
    }
FastLED.show();
  }

  void rainbow()
  {
    // FastLED's built-in rainbow generator
    fill_rainbow( leds, NUM_LEDS, gHue, 8);
    EVERY_N_MILLISECONDS( 20 ) { gHue++; }
    FastLED.show();
  }
```

**The error I get:**
```
CurrentRainbow.ino:7:15: error: variable or field ‘Glitter’ declared void
CurrentRainbow.ino:7:15: error: ‘fract8’ was not declared in this scope
In file included from CurrentRainbow.ino:11:0:
/home/phil/sketchbook/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
                     ^
CurrentRainbow.ino: In function ‘void loop()’:
CurrentRainbow.ino:72:16: error: ‘Glitter’ was not declared in this scope
```

Now I read that fract8 is the same as a byte, and weirdly enough, repacing fract8 with byte successfully compiles.

On my Windows machine using fract8 does not throw that error.

And `fract8` is unsigned wheras `byte` isnt.

kriegsman, 2020-02-12:
Thank you for the Update there at the end!  The script that the Arduino IDE uses to build projects definitely has had problems over the years, and has had a bunch of updates.

fract8 is definitely defined as a type in FastLED, https://github.com/FastLED/FastLED/blob/9ead847a14e7856fc9d1159b31a9598a6e7af89f/lib8tion.h#L336

So if the build process isn't "seeing" it, it probably is related to the order of includes or something similar.  

I'm going to mark this issue as Closed, since there's a fix for the 'fall through' checked in now with https://github.com/FastLED/FastLED/commit/4e5f3251b07a8e4d3aece40d81156ff553ceca5e

If we hit other compiler related issues, we can open a new ticket for them.

Thank you for the help with this!

closed 2020-02-12 by kriegsman