FastLED/FastLED #967: esp32: Flickering occurs when TCP traffic increases

alorbach, 2020-02-11:
Hi all, 

I have a very complex sketch using FastLED with a single LED controller and between 40-255 LEDs. 
My LEDs are WS2813B. 

I am using conditional compilation to build my project for ESP8266 and ESP32 side by side. 
I also use async UDPServer, TCPServer and WebServer Classes from here: 
https://github.com/me-no-dev

Now the problem I am expiriencing only on ESP32 is, that if the TCP traffic increases (Like for example performing an OTA update through the internal Async WebServer), the LED stripes start to get bad flickering. 

- I could reduce the problem somehow by running async TCP classes on internal core 0 (instead of the default core 1). However it is still not fully solved and really would like to know why this is happening and what I can do to avoid it. 

All those precompiled variables don't have any affect as they only work on esp8266 driver
# define FASTLED_ALLOW_INTERRUPTS 1
# define INTERRUPT_THRESHOLD 1
# define FASTLED_INTERRUPT_RETRY_COUNT 0

Any help would be great.

alorbach, 2020-02-11:
Update, it happens with clockless_i2s_esp32 driver primary. 
I tried "clockless_rmt_esp32" as well, but this is totally flickering the whole time.

alorbach, 2020-02-14:
I am so fucking happy, it appears that commit https://github.com/FastLED/FastLED/commit/fec0f0300e14c7dd1fa4fb7eb08ff1fd862f0dae fixed the issue finally :D!
And using this: 
#define FASTLED_ESP32_FLASH_LOCK 1 

Thanks so much @kriegsman

kriegsman, 2020-02-14:
Hey- glad to hear it!  Proper credit goes to @samguyer, and all the people who helped him isolate the "flash lock" situation, and work out a fix for it.

closed 2020-02-14 by kriegsman

samguyer, 2020-02-14:
@alorbach That's great news! Good to get confirmation that this fix works. Here at FastLED we are always shooting for "I am so fucking happy" ;-)

pbolduc, 2020-09-14:
@samguyer does this only relate to SPI flash access? Over on the [WLED](https://github.com/Aircoookie/WLED) project, there have been reports of flickering when using a SPI based DS18B20 temperature sensor. Since this fix was related to SPI it got me thinking. In the WLED side of things the temperature sensor is only being polled every 60 seconds. I haven't had a chance to try to measure any latency involved when accessing the SPI device. The WLED project is still on 3.3.2.