FastLED/FastLED #969: Arduino 33 BLE - ColorPalette.ino doesn't compile

craig24x, 2020-02-13:
Arduino 33 BLE - ColorPalette.ino doesn't compile
error: 'digitPinToBitMask' was nor declared in this scope

/tmp/074269899/custom/FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:63:55: error: 'nrf_pin' is not a member of 'FastPin<5>'

pwm->PSEL.OUT[0] = FastPin<_DATA_PIN>::nrf_pin();

~~~~~~~~~~~~~~~~~~~~~~~~~~~^~

exit status 1

kriegsman, 2020-02-13:
Hi @craig24x and thanks for the report.

Is this the board you're trying to use?  The Arduino Nano 33 BLE?  https://store.arduino.cc/usa/nano-33-ble Or something else?

Do any other FastLED examples compile for that board?  Is it just that one example that's not working for you, or are all of the examples not compiling?

craig24x, 2020-02-14:
Hi, 
 Yes, that link is the exact Arduino. I've tested a few other examples (Fire2012, Pacifica, Pride2015, RGBsetdemo) and all throw the same 'digitPinToBitMask' was nor declared in this scope. 
I assume pin mappings must be different.

zsmatyas, 2020-04-04:
I cannot compile any FastLED examples for my Arduino 33 BLE.

rednebmas, 2020-04-08:
Relevant: https://github.com/FastLED/FastLED/issues/917

rednebmas, 2020-04-08:
Also, for what it's worth, the [Adafruit Neopixel library](https://github.com/adafruit/Adafruit_NeoPixel) _is_ working with my Arduino Nano 33 BLE.

henrygab, 2022-01-18:
To get reliable output on nRF52 chips, it's necessary to use DMA for the output.  This is because the softdevice (SD in Nordic terms) can pre-empt anything else occurring on the CPU (by design) ... and simply enabling BLE is enough to cause problems.

At the time the nRF52 support code was written, only FreeRTOS was used.  As a result, portions of the solution were intrinsically linked to used FreeRTOS.

Is it possible to add support for MBED OS, such as the Nano 33 BLE uses?  Yes.

Is it trivial?  No.

If you have the time and ability to do so, that'll be great! I am not likely to do this in the foreseeable future.