FastLED/FastLED #970: FastLED - Need documented profiling results for common setups.

afaucher, 2020-02-13:
**What happened to me**
I was working on a project for reading incoming IR data (PWM) and updating LEDs based on the data received (laser tag :) ).  My first attempt to do so just had the LEDs update every few thousand cycles on my Arduino.  I would recompute their state and just call FastLED.show().  I noticed however that the timing was severely impacted by the call to FastLED.show() as I was also polling the IR data in the same loop.  I was able to resolve this by just updating LEDs on state change but in the process I did some research and I realized I didn't see any measurements or profiling of FastLEDs in the documentation.

My setup was:
8 NEOPIXEL 5050 LEDs ( I wanted to go to a much longer string of 40ish )
Arduino Duemilanove (Slow :) )

**What I think FastLED should have**
I would love to see a documented profiling for FastLED that looks at common protocols on different devices and tries to measure performance.  Above I was less concerned that it was slow and more concerned that I didn't have an expectation about how slow and what might improve it.

A user of FastLED should be able to answer:
* What performance can I expect on my setup?
* What types of setups give better performance?
* Is my performance expected?
* What type of setup might give me the performance I need?

afaucher, 2020-02-13:
I just found this which is in the direction of what I was looking for:

https://github.com/FastLED/FastLED/wiki/Interrupt-problems

It is more 'you already have this problem' focused and less best practice / recommendation focused.

samguyer, 2020-05-25:
@afaucher Would you like to keep this issue open? Or did the document mostly answer your questions? For specific setups I recommend going on the FastLED Subreddit.

afaucher, 2020-05-26:
@samguyer It answered my specific question but didn't offer the general guidance I was looking for.  I will leave it to you to see if this is something you want to track or not.

I'll summarize the above to simplify the problem.  I would hope to be able to answer: "How can I evaluate a specific micro-controller for use with Fast LED and what resource usage by the library might conflict with other common uses?"