FastLED/FastLED #971: esp32 IDF v4.0

PhilColbert, 2020-02-16:
Anyone got fastled running on the latest esp32 code ?

Thanks

littleblackfish, 2020-02-26:
What would be the most straightforward way? I am trying to follow this with little luck so far. I could build arduino-esp32 as a component, but not FastLED so far. 

https://github.com/espressif/arduino-esp32/blob/idf-release/v4.0/docs/esp-idf_component.md

littleblackfish, 2020-02-27:
After adding a `add_definitions(-DESP32)` on the top CmakeLists.txt file and no other changes, here is where I'm at : 

```
In file included from ../components/arduino/libraries/FastLED/FastLED.h:58,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/colorutils.h: In copy constructor 'CHSVPalette16::CHSVPalette16(const CHSVPalette16&)':
../components/arduino/libraries/FastLED/colorutils.h:455:69: error: 'void* memmove(void*, const void*, size_t)' writing to an object of type 'struct CHSV' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
                                                                     ^
In file included from ../components/arduino/libraries/FastLED/controller.h:9,
                 from ../components/arduino/libraries/FastLED/FastLED.h:47,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/pixeltypes.h:23:8: note: 'struct CHSV' declared here
 struct CHSV {
        ^~~~
In file included from ../components/arduino/libraries/FastLED/FastLED.h:58,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/colorutils.h: In member function 'CHSVPalette16& CHSVPalette16::operator=(const CHSVPalette16&)':
../components/arduino/libraries/FastLED/colorutils.h:459:69: error: 'void* memmove(void*, const void*, size_t)' writing to an object of type 'struct CHSV' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
                                                                     ^
In file included from ../components/arduino/libraries/FastLED/controller.h:9,
                 from ../components/arduino/libraries/FastLED/FastLED.h:47,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/pixeltypes.h:23:8: note: 'struct CHSV' declared here
 struct CHSV {
        ^~~~
In file included from ../components/arduino/libraries/FastLED/FastLED.h:58,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/colorutils.h: In copy constructor 'CHSVPalette256::CHSVPalette256(const CHSVPalette256&)':
../components/arduino/libraries/FastLED/colorutils.h:558:69: error: 'void* memmove(void*, const void*, size_t)' writing to an object of type 'struct CHSV' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
                                                                     ^
In file included from ../components/arduino/libraries/FastLED/controller.h:9,
                 from ../components/arduino/libraries/FastLED/FastLED.h:47,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/pixeltypes.h:23:8: note: 'struct CHSV' declared here
 struct CHSV {
        ^~~~
In file included from ../components/arduino/libraries/FastLED/FastLED.h:58,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/colorutils.h: In member function 'CHSVPalette256& CHSVPalette256::operator=(const CHSVPalette256&)':
../components/arduino/libraries/FastLED/colorutils.h:562:69: error: 'void* memmove(void*, const void*, size_t)' writing to an object of type 'struct CHSV' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
                                                                     ^
In file included from ../components/arduino/libraries/FastLED/controller.h:9,
                 from ../components/arduino/libraries/FastLED/FastLED.h:47,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/pixeltypes.h:23:8: note: 'struct CHSV' declared here
 struct CHSV {
        ^~~~
In file included from ../components/arduino/libraries/FastLED/FastLED.h:58,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/colorutils.h: In copy constructor 'CRGBPalette16::CRGBPalette16(const CRGBPalette16&)':
../components/arduino/libraries/FastLED/colorutils.h:663:69: error: 'void* memmove(void*, const void*, size_t)' writing to an object of type 'struct CRGB' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Werror=class-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
                                                                     ^
In file included from ../components/arduino/libraries/FastLED/controller.h:9,
                 from ../components/arduino/libraries/FastLED/FastLED.h:47,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/pixeltypes.h:102:8: note: 'struct CRGB' declared here
 struct CRGB {
        ^~~~
In file included from ../components/arduino/libraries/FastLED/FastLED.h:58,
[... cut ...] class-memaccess]
         memmove8( &(entries[0]), &(rhs[0]), sizeof( entries));
                                                             ^
In file included from ../components/arduino/libraries/FastLED/controller.h:9,
                 from ../components/arduino/libraries/FastLED/FastLED.h:47,
                 from ../components/arduino/libraries/FastLED/FastLED.cpp:2:
../components/arduino/libraries/FastLED/pixeltypes.h:102:8: note: 'struct CRGB' declared here
 struct CRGB {
        ^~~~
```

littleblackfish, 2020-02-27:
So I replaced all the memmove calls with assignments i.e. 

`memmove8( &(entries[0]), &(rhs[0]), sizeof( entries));`

to 

`entries[0] = rhs[0];`

and it now builds, although I have no idea if it has the correct behavior and have no hardware to test. I can only confirm that the memcopy8 lines in colorutils.h (e.g. https://github.com/FastLED/FastLED/blob/4e5f3251b07a8e4d3aece40d81156ff553ceca5e/colorutils.h#L455) are the only issue keeping this from building with esp-idf v4.0. 

Hopefully someone who knows what they are doing will take a look at this.

littleblackfish, 2020-02-27:
Scratch that, casting all the pointers to (void*), I belive, is the appropriate solution to the problem, i.e. 

`memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));` 

to 

`memmove8( (void*) &(entries[0]), (void*) &(rhs.entries[0]), sizeof( entries));` 
etc. 

This builds without complaints, but I still have no hw to test it. 

Apologies for the confusing messages, my C++ is super rusty. I'd love to do a pull request but I have no way to test this with all the platforms, is there a recommended way to proceed?

bbulkow, 2020-02-28:
Hey, I do, and it works. There's a little hacking going on ( my platform defines aren't right, and I haven't yet debugged to see if I'm using the RMT devices as planned ), but it blinks LEDs properly. This is with the rel4.0 version of esp-idf ( not master, although I think it would probably work ).

Please see my repo at: https://github.com/bbulkow/FastLED-idf

The compile issue with the memmove is actually best done by not using memmove in c++ to copy objects. It's better to do a loop and the object copies. While the optimization of memmove was right for older compilers and older systems, I don't think it's appropriate for gcc8, which is why they "banned" it with warning that, with -Wall, turns into an error. I think that's a better solution than a type cast. I would say that checking the GCC version and doing a straight copy instead of a memmove might be the right choice.

I'll be doing some performance tests etc.

Is there somewhere else I could announce my repo? I don't have a reddit account anymore.

raldone01, 2020-03-06:
There is an issue with the rmt driver. See https://github.com/espressif/esp-idf/issues/4583 and https://github.com/bbulkow/FastLED-idf/pull/2.

5chmidti, 2020-03-28:
I've opened #990 which removes these warning from all uses of memmove in user code and other usage that requires `CRGB` and `CHSV` to be trivially copyable (other discussion was at #985). It removes the need to cast every `memmove` in user code to `void *` when you don't want to have the warnings.
As stated in the pr this could use some performance testing, I've done some testing but I don't know if my tests are any good.

drewandre, 2020-04-01:
I was able to get FastLED v3.3.1 running on esp-idf v4.0 using @5chmidti's 1550a94 commit. I did run into the rmt init issue on esp-idf v4.1 but v4.0 seems to be working great for this library, granted I'm only running noise plus palette. Thanks!