FastLED/FastLED #972: Having trouble compiling for DigiSpark

spierepf, 2020-02-16:
My code works on my Arduino UNO, but when I compile it for my DigiSpark, it balks:

In file included from /home/peter/Arduino/Digispark_Magic_Mushroom/Digispark_Magic_Mushroom.ino:1:0:
/home/peter/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
                     ^
In file included from /home/peter/Arduino/libraries/FastLED/platforms/avr/fastled_avr.h:6:0,
                 from /home/peter/Arduino/libraries/FastLED/platforms.h:39,
                 from /home/peter/Arduino/libraries/FastLED/FastLED.h:52,
                 from /home/peter/Arduino/Digispark_Magic_Mushroom/Digispark_Magic_Mushroom.ino:1:
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showRGBInternal(PixelController<RGB_ORDER>&)':
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:409:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected ';' before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:409:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:410:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected ';' before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:410:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:411:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected ';' before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:411:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:425:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected ';' before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:425:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:426:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected ';' before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:426:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
[... 34 lines cut ...]
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:445:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:446:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:326:24: error: expected ';' before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
/home/peter/Arduino/libraries/FastLED/platforms/avr/clockless_trinket.h:446:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
In file included from /home/peter/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /home/peter/Arduino/Digispark_Magic_Mushroom/Digispark_Magic_Mushroom.ino:1:
/home/peter/Arduino/libraries/FastLED/fastspi.h: At global scope:
/home/peter/Arduino/libraries/FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Multiple libraries were found for "FastLED.h"
 Used: /home/peter/Arduino/libraries/FastLED
exit status 1
Error compiling for board Digispark (16mhz - No USB).

kriegsman, 2020-02-16:
Try pulling the latest 'master' branch from github and using that; there are a couple of changes in there that might fix this for you.  Please let us know, and thanks for the report.

spierepf, 2020-02-17:
I cloned the master branch into my Arduino/libraries directory, and my code compiled. Thanks!

closed 2020-02-17 by spierepf