FastLED/FastLED #978: FastLED wont compile on circuit playground bluefruit

Theacalix, 2020-03-02:
I believe this is similar to [these](https://github.com/FastLED/FastLED/issues/393) [two](https://github.com/FastLED/FastLED/issues/414) older issues, where there are no pin mappings for circuit playground bluefruit since it is a relatively new board. However, I've read through both of those thoroughly and I don't have enough knowledge of microcontrollers or FastLED to fix it myself.  

I've attached the error log with verbose compiler output so you can see the exact errors I am getting. 
[error.txt](https://github.com/FastLED/FastLED/files/4278557/error.txt)

henrygab, 2020-03-09:
Hi @Theacalix,

I put together **[a change]**(https://github.com/henrygab/FastLED/commit/43729537efd6464c15e713b58d4d49a29883afa9#diff-d12b8353741637ac999abc9ea67de13f) that should enable this board.

This is **_untested_** beyond just compiling.  If I get a chance in the next week or two I'll validate all the pins with an oscilliscope, and validate at least the built-in neopixels work.  But, you can try it out now if you like.

Please let me know if this works for you.

For those who want to know, you have to look into the board variant files in the board support package (BSP).  Adafruit does a good job commenting them, you can find them under:
    [`hardware\Adafruit\nrf52\variants\circuitplayground_nrf52840\variant.h`](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/variants/circuitplayground_nrf52840/variant.h)
    [`hardware\Adafruit\nrf52\variants\circuitplayground_nrf52840\variant.cpp`](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/variants/circuitplayground_nrf52840/variant.cpp)

henrygab, 2020-05-30:
Update: PR #1027 is now pending, which fixes a few items from the prior version.

Supported pins:
*  ~~A0 / D12 / AUDIO OUT~~ == connected to on-board audio amplifier, not sure if can output clean signal? 
* [x] A1 / D6
* [x] A2 / D9
* [x] A3 / D10
* [x] A4 / D3 / SCL
* [x] A5 / D2 / SDA
* [x] A6 / D0 / UART RX
* [x] A7 / D1 / UART TX
* [x] D8 -- for the ten built-in Neopixels

PR #1027 (2020-07-29): Enable Circuit playground Bluefruit
This adds the necessary pin definitions for this board.
All pins were toggled using FastPin<>::toggle().

Thanks to @ardnew for pointing out an early error!

Supported pins:
*  ~~A0 / D12 / AUDIO OUT~~ == connected to on-board audio amplifier, not sure if can output clean signal? 
* [x] A1 / D6
* [x] A2 / D9
* [x] A3 / D10
* [x] A4 / D3 / SCL
* [x] A5 / D2 / SDA
* [x] A6 / D0 / UART RX
* [x] A7 / D1 / UART TX
* [x] D8 -- for the ten built-in Neopixels

Manual [test sketch](https://gist.github.com/henrygab/1d4825a72864908e0204c3e18d6ba539) used to verify corr [... cut]
closed 2020-07-29 by kriegsman