FastLED/FastLED #979: no instance of overloaded function "CFastLED::addLeds" matches the argument list

LaVielle, 2020-03-05:
I'm trying to call `FastLED.addLeds` inside a function, and I'm getting error when using a parameter instead of a hard coded value for ledPin:

```C++
void setupLeds(int ledPin, CRGB leds[], int numLeds) {
  FastLED.addLeds<WS2812B, ledPin, RGB>(leds, numLeds);
}
```
In this case I get this error:
```
no instance of overloaded function "CFastLED::addLeds" matches the argument list -- argument types are: (CRGB *, const int) -- object type is: CFastLED
```

If I use a hard coded value for ledPin, it works fine:
```C++
void setupLeds(CRGB leds[], int numLeds) {
  FastLED.addLeds<WS2812B, 9, RGB>(leds, numLeds);
}
```

The param I am passing to my `setupLeds` for ledPin is just `const int someLedPin = 9;`, so the type is what FastLED.addLeds expects, i think.

Am I using FastLED.addLeds the right way?
Am I missing something obvious?

5chmidti, 2020-03-14:
you are using a template with the <>. When using a template the arguments have to be specified at compile time, which means you cant use a simple int as a template argument. You have to give it a value at compile time.
You could use a macro:
```
#define LED_PIN 9

void setupLeds(CRGB leds[], int numLeds) {
  FastLED.addLeds<WS2812B, LED_PIN, RGB>(leds, numLeds);
}
```

samguyer, 2020-05-25:
@5chmidti Thank you!

closed 2020-05-25 by samguyer