FastLED/FastLED #985: FastLED compiler warnings on gcc/linux

marcmerlin, 2020-03-19:
I'm not sure if they are worth fixing/easy to fix. F
[out.txt](https://github.com/FastLED/FastLED/files/4356359/out.txt)
ull list attached.
```
libraries/FastLED/colorutils.h: In copy constructor ‘CHSVPalette16::CHSVPalette16(const CHSVPalette16&)’:
libraries/FastLED/colorutils.h:455:69: warning: ‘void* memmove(void*, const void*, size_t)’ writing to an object of type ‘struct CHSV’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
                                                                     ^
In file included from libraries/FastLED/controller.h:9,
                 from libraries/FastLED/FastLED.h:47,
                 from examples/GFX_Table_Mark_Estes/neomatrix_config.h:112,
                 from examples/GFX_Table_Mark_Estes/Table_Mark_Estes.ino:5,
                 from src/main.cpp:13:
libraries/FastLED/pixeltypes.h:23:8: note: ‘struct CHSV’ declared here
 struct CHSV {
        ^~~~
In file included from libraries/FastLED/FastLED.h:58,
                 from examples/GFX_Table_Mark_Estes/neomatrix_config.h:112,
                 from examples/GFX_Table_Mark_Estes/Table_Mark_Estes.ino:5,
                 from src/main.cpp:13:
libraries/FastLED/colorutils.h: In member function ‘CHSVPalette16& CHSVPalette16::operator=(const CHSVPalette16&)’:
libraries/FastLED/colorutils.h:459:69: warning: ‘void* memmove(void*, const void*, size_t)’ writing to an object of type ‘struct CHSV’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
                                                                     ^
In file included from libraries/FastLED/controller.h:9,
                 from libraries/FastLED/FastLED.h:47,
                 from examples/GFX_Table_Mark_Estes/neomatrix_config.h:112,
                 from examples/GFX_Table_Mark_Estes/Table_Mark_Estes.ino:5,
                 from src/main.cpp:13:
libraries/FastLED/pixeltypes.h:23:8: note: ‘struct CHSV’ declared here
 struct CHSV {
        ^~~~
In file included from libraries/FastLED/FastLED.h:58,
                 from examples/GFX_Table_Mark_Estes/neomatrix_config.h:112,
                 from examples/GFX_Table_Mark_Estes/Table_Mark_Estes.ino:5,
                 from src/main.cpp:13:
libraries/FastLED/colorutils.h: In copy constructor ‘CHSVPalette256::CHSVPalette256(const CHSVPalette256&)’:
libraries/FastLED/colorutils.h:558:69: warning: ‘void* memmove(void*, const void*, size_t)’ writing to an object of type ‘struct CHSV’ with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
         memmove8( &(entries[0]), &(rhs.entries[0]), sizeof( entries));
```

5chmidti, 2020-03-21:
How about casting to a `void*` to get around it being recognized as a `struct CHSV` in `memmove8`. Because the copy constructor of `CHSVPalette16` is using `memmove` anyways, there is no need to have it complain about not having a trivial copy constructor. (I guess this is true for all CHSVPalettes)
Would need to cast at ln 455 and 459 to test this. I cant, as I would need a newer gcc version

marcmerlin, 2020-03-24:
I cherry picked this into master: https://github.com/FastLED/FastLED/pull/986
Not sure who has review power outside of @kriegsman , do you by any chance?

5chmidti, 2020-03-25:
I believe everybody can review changes in a pull request. Merging is reserved for people with write permission.

I looked at it once more and wrote something in the pr to completely remove the warning instead of going around it.

5chmidti, 2020-03-27:
As I've said in the pr (should have stayed here, to keep the discussion in here):
> I've done a quick test on the following (only if it compiles on c++14, but this should be correct according to the standard):
> Why not just outright remove the warnings about `CRGB` and `CHSV` not being trivially copyable instead of casting the warning away?
> By simply removing the user defined copy constructor' and copy assignments', `CRGB` and `CHSV` could simply be made trivially copyable by using the default copy constructor and copy assignment. One could write explicitly that we want to use the default with for example:
> `inline CRGB &operator=(const CRGB &rhs) __attribute__((always_inline)) = default;`
> Which I guess could be preferrable or even necessary because we have other copy assignments with other arguments for `CRGB`, as I don't know if the copy assignment for `CRGB` would be deleted.
> After that the warnings shouldn't be there anymore, as the types would be trivially copyable.

I've gone ahead and commited these changes to my [fork](https://github.com/5chmidti/FastLED/commit/1550a942ba69272e10e4ca56fd51dd9f074e1671).

Performance wise there shouldn't be a difference between this and the master, but if someone knows how to reliably check, please do so. I've done some testing myself and found that they are about the same (no major difference / within margin of error).

5chmidti, 2020-03-28:
I've opened #990

marcmerlin, 2020-03-28:
Thanks for following up on this (been busy/distracted with other things)

zackees, 2026-08-20:
Resolved by #986 and #990. Verified: both PRs merged to master; #986 removed the reported palette memmove warning sites and #990 made CRGB/CHSV copy construction and assignment trivial, which the current implementation retains.

closed 2026-08-20 by zackees