FastLED/FastLED #988: My issue: howto Control multiple Strips in one array with reversed Order of one strip

wildblue, 2020-03-26:
Please Help me, because i can Not find any information regarding the Control of one Strip in reversed Order and the Solution i think of is Not working.

I want to have one Array of Pixels, with multiple Strips INSIDE, wich are connected to different Pins of my Arduino Uno. I an using ws2812 Strips.

I found, that you can Reverse the Order in a crgbset, but i could Not get this to Work.

My Layout ist as follows:
CRGBSet leds_bottom (leds(0, 35-1));
CRGBSet leds_left (leds(35, 57-1));
Crgbset leds_right (leds(159-1, 97)); //reversed
Crgbset leds_back (leds(159, 297-1)); 

I then had in mind to build a new array with the Pixels from above and to use the CRGBSets individually  but i am stuck at this step, because the definition in Reverse does Not Work as expected and i am getting Strange Results.

Please Help me Out in this.

wildblue, 2020-03-26:
I think the issue is in pixelset Line 37 and Line 45:

37: inline CPixelView(PIXEL_TYPE *_leds, int _start, int _end) : dir(((_end-_start)<0) ? -1 : 1), len((_end - _start) + dir), leds(_leds + _start), end_pos(_leds + _start + len) {}

45: bool reversed() { return len < 0; }

This is the constructor and its helper i try to use and i think there is the issue. The length  for reversed Strips only gets Set correctly when  but the reverse direction only gets set correct, when the end Pin

marmilicious, 2020-03-27:
CRGBSet needs continuous ranges, so your (0, 35-1) won't work, but (35-0) should.

We use github 'issues' here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group: https://www.reddit.com/r/FastLED
Please note group rules when sharing code there.

wildblue, 2021-03-27:
I know for what GitHub is intended, as i am a Software Developer for many years. But this is the only place where i thought this has to be, as the dicumentation could not help me and the API behaves in a non working way.

So do you mean
   CRGBSet leds_bottom (leds(35-0, 0));
or
   CRGBSet leds_bottom (leds(35, 0));
or
   CRGBSet leds_bottom (leds(34, 0));

Remember, that i want to have the pixel order inverted and that this array has 36 elements.

Thanks for your time in advance, i am very happy to work on this issue again, as it seems i now have a solution for my intended usecase. By the way, this Project is awesome...

marmilicious, 2021-03-28:
Sorry for the confusion.  My post from a year ago should have used a comma like this.  
```
CRGBSet leds_bottom(leds(35, 0));
```

zackees, 2026-08-20:
Resolved by #1756. Verified: the merged master implementation preserves reverse CPixelView ordering; current pixelset.h documents inclusive reversed subsets such as pixels(50, 10), and the focused pixelset regression suite passes.

closed 2026-08-20 by zackees