FastLED/FastLED #989: Red and green inversed

marcounetB, 2020-03-26:
Hello !
For both HSV and RGB systems, red and green color are switched !
i am on the last version of the library and have no idea where it could come from.

marmilicious, 2020-03-27:
Have you run the RGBCalibrate example that comes with the library to check the color for your strip?
https://github.com/FastLED/FastLED/blob/master/examples/RGBCalibrate/RGBCalibrate.ino

If this doesn't get you sorted out please post a question (with details of your setup) on the FastLED Reddit group.

MarcBresson, 2023-08-28:
I am closing that, I will do further testing because in my memory, the problem was still happening. If I ever open a new issue, I ll provide more details as to how to reproduce the problem.

Thank you for taking the project back, and good luck.

closed 2023-08-28 by MarcBresson