FastLED/FastLED #999: FastLed Esp32 Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)

JonnyGree, 2020-04-18:
Hello, 

I have this code for control a stepper motor and a led trought EspNow. I try to summarize it for explain the problem.

1 - In main i recive data trought EspNow
   `void OnDataRecv()`
... after some calculations 
2- call  MotorON() that  turn led red and enable interrupt 
`void MotorON(){`  
        `leds[0] = CRGB::Red; `
        `FastLED.show(); `
 ` timer2 = timerBegin(2, 16, true); ` 
`  timerAttachInterrupt(timer2, &onTimer2, true); `
  `timerAlarmWrite(timer2, 100, true); `
`  timerAlarmEnable(timer2); `
`}`

3- do interrupt subroutine 
`void IRAM_ATTR onTimer2() {`
  `portENTER_CRITICAL_ISR(&timerMux2);`
`        ...`      
`  portEXIT_CRITICAL_ISR(&timerMux2);  `
`   if (condition ) MotorOFF(); `
`}`
4-when the condition is satisfied i call MotorOFF(); that turn off ISR and turn led green.
`void MotorOFF(){`
`   timerEnd(timer2);`
`   timer2 = NULL;  `
`         leds[0] = CRGB::Green; `
`         //uncomment cause CRASH`
`         //FastLED.show();`
`} `

At this point if i uncomment the line FastLED.show(); i got this error
    Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)

if it can help,  in point 3 if (condition ) MotorOFF(); is put befor the port exit it get the error in both core

I Attach the code that run after 8 second this for point once. 

[Stepper_Control.zip](https://github.com/FastLED/FastLED/files/4496577/Stepper_Control.zip)

DO somebody know how to fix it ?

sanvd0000, 2020-06-24:
I have the same problem, tried: #define FASTLED_ALLOW_INTERRUPTS 0
didnt work though.

sanvd0000, 2020-06-24:
https://github.com/FastLED/FastLED/wiki/Interrupt-problems look at this!

szantogab, 2023-11-25:
@sanvd0000 have you managed to figure out this one ? I'm having the same issue on an ESP32-C3. Thanks!

Aayushyamaan-Shah, 2023-12-28:
> @sanvd0000 have you managed to figure out this one ? I'm having the same issue on an ESP32-C3. Thanks!

I am facing the same issue. I tried disabling intrrupts, interrupt retrys and many other methods. It did not work.

szantogab, 2024-05-14:
If I try to call FastLED.show() as few times as possible, then this error goes away. But still it's not a good solution in my case. Any ideas? The same code works on an WeMos D1 mini.

closed 2026-08-20 by zackees