FastLED/FastLED #1002: fill_gradient_RGB with 3 colors causing memory issue?

marmilicious, 2020-04-19:
Using fill_gradient_RGB with three (3) colors usually causes the controller to continuously reboot.

This was reported in the Reddit group:
https://www.reddit.com/r/FastLED/comments/g4fow5/fill_gradient_rgb_with_3_colours/

I tested a bit and confirmed that using fill_gradient_RGB with three colors seems to be causing something strange in memory.  Here's my test file with comments of what I was observing with using different colors.

https://pastebin.com/kv95Z6UV

marmilicious, 2020-04-20:
Forgot to note, sometimes the compiler will give this warning when uploading:
`warning: large integer implicitly truncated to unsigned type 
`

5chmidti, 2020-04-20:
The underlying problem would be that `CRGB::Red` does not mean, that it is a CRGB object.
In `void fill_gradient_RGB( CRGB* leds, uint16_t numLeds, const CRGB& c1, const CRGB& c2);`
calling `fill_gradient_RGB(leds, NUM_LEDS, CRGB::Red, CRGB::Blue);` would implicitly convert the enums into CRGB objects via `    inline CRGB( uint32_t colorcode) _attribute__((always_inline))`, because that's the only way that call makes sense. 
But when you want to use the three color variant the compiler selects `void fill_gradient_RGB( CRGB* leds,  uint16_t startpos, CRGB startcolor, uint16_t endpos,   CRGB endcolor );` over `void fill_gradient_RGB( CRGB* leds, uint16_t numLeds, const CRGB& c1, const CRGB& c2, const CRGB& c3);` when using the predefined enum colors, because it is cheaper to convert the enum to a uint16_t/literal type than to a struct.

This problem however only applies to using the predefined colors of the enum in CRGB. Using `fill_gradient_RGB(leds, NUM_LEDS, CRGB{255, 0, 0}, CRGB{64, 128, 0}, CRGB{0, 255, 0});` works fine for me.

`warning: large integer implicitly truncated to unsigned type` could be removed by making the underlying type of the enum uin32_t.

marmilicious, 2020-04-22:
Thank you for explaining @5chmidti !

closed 2020-11-17 by marmilicious