FastLED/FastLED #1009: One CRGB[], multiple pins with multiple color order broke since last update (ESP32?)

Insomniac80, 2020-04-30:
Hi,

I have a strange behaviour since last update. Using ESP32, dunno if it is also on other platforms. Worked fine on 3.003.002, is broken on 3.003.003.

When adding different multiple stripes on different pins, which have different color orders, there are two problems.
My setup is: 
6 pins (13,14,16,17,18,19)
2 different stripe types (2xWS2811 & 4xWS2812B)
which have three different color orders (2xBRG, 1x GRB, 3xRGB). 
I have one CRGB-Array, then add the stripes to this with the different properties, as usual:

```
FastLED.addLeds<STRIPE_TYPE_01, LED_PIN_01, COLOR_MODE_01>(leds, 0, NUM_LEDS_01);
FastLED.addLeds<STRIPE_TYPE_02, LED_PIN_02, COLOR_MODE_02>(leds, NUM_LEDS_01, NUM_LEDS_02);
[...]
```

Problem is, all stripes are using the color order (f.e. BRG) handed over for the first stripe. Also, the first two LEDs of the first stripe also show a different color then the other LEDs of the same stripe when setting all to the same color.
As soon as I go back to 3.003.002, it works fine again.

jordanadania, 2020-06-07:
I just made a 2nd CRGB[] for the different color order

agrath, 2020-09-15:
I think this may be the underlying issue https://github.com/FastLED/FastLED/issues/827

closed 2026-08-20 by zackees