FastLED/FastLED #1011: does this Library support WS2818 chip?

gregcrago, 2020-05-02:
I have read it is similar to thw WS2811 but with redundant data to allow for a 'dead' pixels and still light downstream pixrels. If this library does support WS2818, can someone please add WS2818 to supported chip list?

marmilicious, 2020-05-02:
Did you try specifying the type as WS2811?  Sounds like it should work.
https://github.com/FastLED/FastLED/issues/468

If your question is more about "can it be added so it's a valid chipset type to put in the addLeds line?" then yes, I imagine at some point it will be added.

If you have a WS2818 strip, can you report back if specifying WS2811 works for you?

marmilicious, 2020-05-02:
@gregcrago  Or maybe try using WS2813?  Please let us know if you try it.
https://www.reddit.com/r/FastLED/comments/e0szhe/ws2818_string_will_not_light/

gregcrago, 2020-05-03:
I was thinking about purchasing a set of WS2818 Strip, but I wanted to know if FastLED could drive these. No, I do not have these lights

closed 2026-08-20 by zackees