FastLED/FastLED #1014: ColorFromPalette in 16 bits?

RomRider, 2020-05-08:
Hi,

I always love to use FastLED with most of my arduino projects, best library ever :)

I'm wondering lately about the possibility to have a ColorFromPalette in 16bits definition (`index` in 16bits) in addition to the 8bit definition that is currently in place in order to have smoother transitions? Right now, if you have very slow transitions in place throughout the palette, you can see the (very small) "jumps" from one color to the other on the LEDs.

Something like:
```cpp
CRGB ColorFromPalette16(const CRGBPalette16& pal,
                        uint16_t index,
                        uint8_t brightness=255,
                        TBlendType blendType=LINEARBLEND);
```

I understand that it would be more compute intensive though.

5chmidti, 2020-05-08:
See #202 . Been a pr for some time now, who knows when it will get merged. But you could try it out by maybe making a custom local branch with that commit added if you want.

RomRider, 2020-05-09:
Thanks @5chmidti !
I've added this in my code directly. It mostly works but some things where overflowing see below:

```cpp
CRGB sevenSegments::colorFromPaletteExtended(const CRGBPalette16 &pal, uint16_t index, uint8_t brightness, TBlendType blendType)
{
    // This fucnction has the same intuitive behavior as the other ColorFromPalette
    // functions, except it provides 8-bit interpolation between palette entries.

    // Extract the four most significant bits of the index as a palette index.
    uint8_t index_4bit = (index >> 12);

    // Calculate the 8-bit offset from the palette index.
    // Throws away the 4 least significant bits and uses the middle 8.
    uint8_t offset = (uint8_t)(index >> 4);

    // Get the palette entry from the 4-bit index
    //  CRGB rgb1 = pal[ hi4];
    const CRGB *entry = &(pal[0]) + index_4bit;
    uint8_t red1 = entry->red;
    uint8_t green1 = entry->green;
    uint8_t blue1 = entry->blue;

    uint8_t blend = offset && (blendType != NOBLEND);

    if (blend)
    {
        // If palette blending is enabled, use the offset to interpolate between
        // the selected palette entry and the next.

        if (index_4bit == 15)
        {
            entry = &(pal[0]);
        }
        else
        {
            entry++;
        }

        // Calculate the scaling factor and scaled values for the lower palette value.

        uint8_t f1 = 256 - offset;

        red1 = scale8_LEAVING_R1_DIRTY(red1, f1);
        green1 = scale8_LEAVING_R1_DIRTY(green1, f1);
        blue1 = scale8_LEAVING_R1_DIRTY(blue1, f1);

        // Calculate the scaled values for the neighboring palette value.
        uint8_t red2 = entry->red;
        uint8_t green2 = entry->green;
        uint8_t blue2 = entry->blue;
        red2 = scale8_LEAVING_R1_DIRTY(red2, offset);
        green2 = scale8_LEAVING_R1_DIRTY(green2, offset);
        blue2 = scale8_LEAVING_R1_DIRTY(blue2, offset);

        cleanup_R1();

        // This was overflowing so I've updated it using qadd8
        red1 = qadd8(red1, red2);
        green1 = qadd8(green1, green2);
        blue1 = qadd8(blue1, blue2);
    }

    if (brightness != 255)
    {
        nscale8x3_video(red1, green1, blue1, brightness);
    }

    return CRGB(red1, green1, blue1);
}

```

leifclaesson, 2020-12-06:
@RomRider Thanks for this!

Here it is modified for CRGBPalette32.

```
CRGB ColorFromPaletteExtended(const CRGBPalette32 &pal, uint16_t index, uint8_t brightness, TBlendType blendType)
{
    // This function has the same intuitive behavior as the other ColorFromPalette
    // functions, except it provides 8-bit interpolation between palette entries.

    // Extract the five most significant bits of the index as a palette index.
    uint8_t index_5bit = (index >> 11);

    // Calculate the 8-bit offset from the palette index.
    // Throws away the 3 least significant bits and uses the next 8.
    uint8_t offset = (uint8_t)(index >> 3);

    // Get the palette entry from the 5-bit index
    const CRGB *entry = &(pal[0]) + index_5bit;
    uint8_t red1 = entry->red;
    uint8_t green1 = entry->green;
    uint8_t blue1 = entry->blue;

    uint8_t blend = offset && (blendType != NOBLEND);

    if (blend)
    {
        // If palette blending is enabled, use the offset to interpolate between
        // the selected palette entry and the next.

        if (index_5bit == 31)
        {
            entry = &(pal[0]);
        }
        else
        {
            entry++;
        }

        // Calculate the scaling factor and scaled values for the lower palette value.

        uint8_t f1 = 256 - offset;

        red1 = scale8_LEAVING_R1_DIRTY(red1, f1);
        green1 = scale8_LEAVING_R1_DIRTY(green1, f1);
        blue1 = scale8_LEAVING_R1_DIRTY(blue1, f1);

        // Calculate the scaled values for the neighboring palette value.
        uint8_t red2 = entry->red;
        uint8_t green2 = entry->green;
        uint8_t blue2 = entry->blue;
        red2 = scale8_LEAVING_R1_DIRTY(red2, offset);
        green2 = scale8_LEAVING_R1_DIRTY(green2, offset);
        blue2 = scale8_LEAVING_R1_DIRTY(blue2, offset);

        cleanup_R1();

        // This was overflowing so I've updated it using qadd8
        red1 = qadd8(red1, red2);
        green1 = qadd8(green1, green2);
        blue1 = qadd8(blue1, blue2);
    }

    if (brightness != 255)
    {
        nscale8x3_video(red1, green1, blue1, brightness);
    }

    return CRGB(red1, green1, blue1);
}
```

sutaburosu, 2020-12-31:
>         // Calculate the scaling factor and scaled values for the lower palette value.
> 
>         uint8_t f1 = 256 - offset;

There's that bug again.  `256` should be `255`.

zackees, 2026-08-20:
Resolved by #1687. Verified on current master: ColorFromPaletteExtended(const CRGBPalette16&, fl::u16, ...) provides the requested 16-bit-index palette lookup, with regression coverage retained in 	ests/fl/gfx/colorutils.cpp; #2373 additionally added the CRGB16 high-precision output path.

closed 2026-08-20 by zackees