FastLED/FastLED #1017: FastLED.clear() and FastLED.clearData() clear only the first LED strip

NiekLambert, 2020-05-15:
Minor issue, easily worked around and probably easily fixed in the library: I am controlling four WS2812 LED strips from a Teensy4. The Arduino 1.8.12 environment with Teensyduino 1.51 reports "FastLED version 3.003.001". Everything works like a charm but when I issue FastLED.clear() or FastLED.clear(true) or FastLED.clearData() only the first LED strip gets cleared which is probably not intended.

MartyMacGyver, 2020-05-25:
@NiekLambert What is your code doing immediately before and after the `clear()` call? I can reproduce this by calling:

```
  LEDS.clear(true);
  FastLED.delay(1000);
```

This is notable as `clear()` is currently:

```
void CFastLED::clear(bool writeData) {
	if(writeData) {
		showColor(CRGB(0,0,0), 0);
	}
    clearData();
}
```

So I see a brief flicker as the strip sets to all black, then immediately switches back to whatever color it was previously. If I reduce the refresh rate down below 10, it's even more obvious.

What catches my eye is the call to `show()` that's integral to `delay()`'s busy loop... I don't understand the purpose of that, but it certainly has an impact (however, on the Teensy, removing it (while still ensuring a normal call to `FastLED.show();` in the example code) causes some oddness when the Serial port is connected - again).

Under all this, though, are calls to `millis()` and `micros()` - I have some suspicion that there's something about Teensyduino's implementation of these (and Serial as well) that's leading to side-effects. It's just a hunch for now, but I'm seeing things I can't explain purely with FastLED code. (For instance, using regular `delay()` instead of `FastLED.delay()` can cause odd behavior if the Serial port is connected). I think the common denominator may be interrupts.

Will update as I learn more.

MartyMacGyver, 2020-05-26:
Well, I can see what the root cause is now in the library - not yet sure how to resolve it though.

Given numLanes and numLeds, you need to create an `leds` array thus:

`CRGB leds[numLanes * numLeds];`

When you use the `FastLED.addLeds<numLanes, WS2812B, DATA_PIN, RGB>(leds, numLeds);` it creates a controller with `numLanes` and `numLeds`. 

Each controller knows its own size, BUT `clear()` calls `clearData()` which calls `clearLedData()` which finally calls

`memset8((void*)m_Data, 0, sizeof(struct CRGB) * m_nLeds);`

The problem is that `m_nLeds` == `numLeds` above, as this is set in `addLeds()`. I wonder if any other platforms have this problem or just Teensy 4.x? I think usually one says "here's the array and here's the data pin to manage", not "here's the full array for all the leds, the number of lanes, and the starting pin to a predefined list of pins valid for this construction". The latter shortcut construction is managed as a `FlexibleInlineBlockClocklessController`, entirely new for Teensy 4.x.

Normally, you could just deal with the controllers as separate entities (e.g, `FastLED[0]`, `FastLED[1]`, etc.) but in this case it's one controller with lots of LEDs grouped together in parallel (but in one long memory array). 

I'm going to think about ways to fix this that don't break anything else (though they may mean that Teensy 4.x sketches would need minor adjustments going forward). The simplest thing is to give the *total* number of LEDs to addLeds and then slice it from there within the `FlexibleInlineBlockClocklessController`.

More updates as this progresses.

NiekLambert, 2020-05-26:
My code is complicated and consists of many files. I currently clear all
the leds in a for loop and thus avoid using clear() and clearData. I
currently operate 4 adjacent 16*16 led matrices. Using a single controller
with num_lanes = 4 and 4 data pins keeps the code simple and the update
speed pleasant: FastLED.show() now takes about 8ms. I wasn't aware that
this is a new feature specific for the Teensy4.
BTW The 8ms of FastLED.show() actually dominates performance completely on
the Teensy4: even doing fancy things like evaluating floating
point trancendental functions for each LED pixel happens in a fraction of
this time.

I typically do use a brief delay() immediately after FastLED.show(), and I
do initialize the Serial port (for some very minor state logging and/or
debugging). In some cases I see spectacularly colorful flickering e.g. when
I intentionally try to briefly blacken the leds via
FastLED.setBrightness(0); FastLED.show(); delay(100); and back to normal
brightness. Odd, but no problem for me. Do you think these artefacts should
disappear without delay() or when the Serial port is not initialized?

MartyMacGyver, 2020-05-27:
`clear()` will take longer to fix.

One thing that ought to help is to dial down the refresh rate... for your 4x256 setup, I'd suggest `FastLED.setMaxRefreshRate(120);` or less. If you use that already, what is it set to? And what are the actual values on your addLeds() line? It ought to be something like:

`FastLED.addLeds<4, WS2812B, {starting_pin}, GRB>(leds, 256); `

I presume you also have 4 and 256 (or variables with the same values) currently as shown?

NiekLambert, 2020-05-27:
When I filed the issue I assumed it was a simple matter of something like
using numLanes * m_nLeds to clear all the leds with default value  numLanes
=1 for the regular single strip.case. But apparently you cannot touch that
part of the library and have to work around it?

Filling in all macro constants I indeed use
FastLED.addLeds<4, WS2812, 19, GRB>(leds, 256);
I have not used FastLED.setMaxRefreshRate(), but the FastLED.show() timing
I got was slightly below 8ms so the max refresh rate is indeed close to
120Hz. I typically call FastLED.show() close to full speed to achieve
optimum dithering for low led intensities and do timing of the dynamics via
dependency on millis().

MartyMacGyver, 2020-05-28:
If you call setMaxRefreshRate() it should help with flickering.

As for the fix, the problem is that the Teensy 4.x code is very new and somewhat unique versus how FastLED normally is used. Either we need to fix all the functions that aren't getting the right number of LEDs or we find a clever way of ensuring the right data gets to those functions in the first place. I'm looking at the latter, but it raises questions of user experience expectations, thus my questions.

Once I have made some progress I'll follow up.

board707, 2021-03-07:
Hi, Martin!
I suppose the problem with clear() can be easily solved ...
As you wrote above - "each controller knows its own size", so we can replace
`memset8((void*)m_Data, 0, sizeof(struct CRGB) * m_nLeds);`
with
`memset8((void*)m_Data, 0, sizeof(struct CRGB) * size() );`
 in clearLedData() code, because size() is virtual method, specific to each controller.
It seems to me that this will fix the problem and will not affect anything else.