FastLED/FastLED #1022: WS2813 with NodeMCU ESP8266 12E display strip is static

baqwas, 2020-05-26:
Hello,

I cloned the current master onto Ubuntu 20.04 and driving the Alitove **WS2813** with Alitove 20A power supply with an NodeMCU **ESP8266** 12E board. My changes to the standard examples are:

1st line:
`#define FASTLED_ESP8266_NODEMCU_PIN_ORDER`

#define changes:
`#define DATA_PIN            D3
#define NUM_LEDS            300
#define MAX_POWER_MILLIAMPS 60
#define LED_TYPE            WS2813
#define COLOR_ORDER         GRB`

Approx. 30 LEDS are lit irrespective of the setting for NUM_LEDS and the LEDs remain static after uploading is complete (the LEDs blink during upload). I seem to have challenges with the DATA_PIN but settled on D3 for the time being. There is nothing else connected to the NodeMCU board.

The strip wire connections are: 
* Red, 5V (from power supply)
* Green, BI (unused), did connect jointly with Yellow but that made no difference
* Yellow, DI, signal from DATA_PIN
* Black, GND, connected to NodeMCU GND

I do have a nominal capacitor between the NodeMCU board's 3V3 and GND pins aligned with correct polarity. I do not have any external pull-up resistor (because I don't know where to use one in the current round of tests).

The GND for the strip and NodeMCU are connected but presently I have omitted the GND from the AC power end (assuming that there is internal common GND in the power supply cage).

All the examples behave in the same manner (blink during upload, static after upload) leading me to believe that I have made some basic/fundamental mistake(s) in the wiring or #define statements. However, since the wiring is minimal I am at a loss to understand the issue. The power supply (at 60 mA per LED) is sufficient yet there is no data at the opposite end of the connected strip.

Any guidance will be sincerely appreciated since the FastLED library does support WS2813 based on no compiler warning/error when LED_TYPE is set appropriately. I don't think it is an issue but of a beginner/_newbie_/user/operator configuration error. Thanks for your understanding.

Kind regards.

marmilicious, 2020-05-28:
5V and GND from the power supply should go directly to the strip.  And then there needs to be a common ground connection from the power supply GND or strip GND (wherever is convenient) to the controller GND.

BI on the first pixel can be connected to GND as well.

We use github "issues" here mostly for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

baqwas, 2020-05-28:
Hello @marmilicious,
My apologies for posting inappropriately. I didn't where to turn. I had previously done exactly what you suggested but had no luck. Obviously, some operator/user issue and I had concerns about posting here but in my predicament I was at a loss. Once again, sorry.
Kind regards.

closed 2020-05-28 by baqwas

marmilicious, 2020-05-28:
@baqwas No worries.  If you don't have it working yet, post on the reddit group and hopefully some of the many people there can help get you going.  Photos or sketches of your wiring can be useful to help trouble shoot if you post there.  Please use pastebin.com or gist.github.com and share a link to your code.