FastLED/FastLED #1028: Compilation Error

kp1851, 2020-05-28:
Compilation Error in PlatformIO

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/nanoatmega328.html
PLATFORM: Atmel AVR 2.1.0 > Arduino Nano ATmega328
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 30KB Flash
DEBUG: Current (simavr) On-board (simavr)
PACKAGES:
 - framework-arduino-avr 5.0.0 
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <FastLED> 3.3.3
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
Building in release mode
Compiling .pio\build\nanoatmega328\lib1d9\FastLED_ID126\FastLED.cpp.o
Compiling .pio\build\nanoatmega328\lib1d9\FastLED_ID126\bitswap.cpp.o
Compiling .pio\build\nanoatmega328\lib1d9\FastLED_ID126\colorpalettes.cpp.o
Compiling .pio\build\nanoatmega328\lib1d9\FastLED_ID126\colorutils.cpp.o
In file included from C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/fastled_avr.h:6:0,
                 from C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms.h:39,
                 from C:\Users\kpgir\.platformio\lib\FastLED_ID126\FastLED.h:52,
                 from C:\Users\kpgir\.platformio\lib\FastLED_ID126\FastLED.cpp:2:
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showRGBInternal(PixelController<RGB_ORDER>&)':
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:409:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:410:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:411:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:425:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:426:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:427:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
In file included from C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/fastled_avr.h:6:0,        
                 from C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms.h:39,
                 from C:\Users\kpgir\.platformio\lib\FastLED_ID126\FastLED.h:52,
                 from C:\Users\kpgir\.platformio\lib\FastLED_ID126\colorpalettes.cpp:4:
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h: In static member function 'static void ClocklessController<DATA_PIN, T1, T2, T3, RGB_ORDER, XTRA0, FLIP, WAIT_TIME>::showRGBInternal(PixelController<RGB_ORDER>&)':
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
[... 169 lines cut ...]
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:427:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:444:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 4: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:445:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 3: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
*** [.pio\build\nanoatmega328\lib1d9\FastLED_ID126\colorpalettes.cpp.o] Error 1
*** [.pio\build\nanoatmega328\lib1d9\FastLED_ID126\bitswap.cpp.o] Error 1
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:326:24: error: expected primary-expression before '__attribute__'
 #define FL_FALLTHROUGH __attribute__ ((fallthrough));
                        ^
C:\Users\kpgir\.platformio\lib\FastLED_ID126\platforms/avr/clockless_trinket.h:446:55: note: in expansion of macro 'FL_FALLTHROUGH'
      case 2: _D2(0) LO1 _D3(0) HI1 _D1(1) QLO2(b0,0)  FL_FALLTHROUGH
                                                       ^
*** [.pio\build\nanoatmega328\lib1d9\FastLED_ID126\colorutils.cpp.o] Error 1
======================================= [FAILED] Took 2.34 seconds =======================================

5chmidti, 2020-05-29:
Fixed as of 4e5f3251b07a8e4d3aece40d81156ff553ceca5e, which is not part of a release, so you'd have to clone the repo with git or manually reproduce the changes in the file.

kp1851, 2020-05-30:
I solved this problem by installing ave-gcc 9.2.0 in Arduino IDE
https://blog.zakkemble.net/avr-gcc-builds/.
Sorry guys

closed 2020-05-30 by kp1851