FastLED/FastLED #1029: FastLED support for Adafruit PIXIE on ESP8266/32

highoctane305, 2020-05-29:
It seems I cannot compile something as simple as the FastLED Blink sketch on ESP8266 if I use a version of the esp8266 support library from espressif that is NEWER than version 2.5.2. It doesn't matter if I compile for a generic ESP8266 module or the NodeMCU 1.0 (which is the hardware I actually have).

The line of code that causes the error is:
`FastLED.addLeds<PIXIE, DATA_PIN, RGB>(leds, NUM_LEDS);`

And the bottom few lines of the error I get from the Arduino IDE is this:
`In file included from /tmp/arduino_modified_sketch_352186/Blink.ino:2:0:
/home/steven/Arduino/libraries/FastLED/FastLED.h:431:25: note: template<EBlockChipsets CHIPSET, int NUM_LANES, EOrder RGB_ORDER> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^
/home/steven/Arduino/libraries/FastLED/FastLED.h:431:25: note:   template argument deduction/substitution failed:
Blink:11:53: error: template argument 1 is invalid
      FastLED.addLeds<PIXIE, DATA_PIN>(leds, NUM_LEDS);
                                                     ^
In file included from /tmp/arduino_modified_sketch_352186/Blink.ino:2:0:
/home/steven/Arduino/libraries/FastLED/FastLED.h:472:25: note: template<EBlockChipsets CHIPSET, int NUM_LANES> static CLEDController& CFastLED::addLeds(CRGB*, int, int)
  static CLEDController &addLeds(struct CRGB *data, int nLedsOrOffset, int nLedsIfOffset = 0) {
                         ^
/home/steven/Arduino/libraries/FastLED/FastLED.h:472:25: note:   template argument deduction/substitution failed:
Blink:11:53: error: template argument 1 is invalid
      FastLED.addLeds<PIXIE, DATA_PIN>(leds, NUM_LEDS);
                                                     ^
exit status 1
'PIXIE' was not declared in this scope'

I've been using the PIXIEs a lot, and as I say, this problem only shows up with version of ESP8266 later than 2.5.2. Am I doing something wrong or is this a bug?

Thanks,
Steven

5chmidti, 2020-05-30:
See [here](https://github.com/FastLED/FastLED/wiki/Chipset-reference):
> Adafruit Pixie leds (PIXIE) - http://www.adafruit.com/product/2741 - in order to use these leds you #include <SoftwareSerial.h> before #include <FastLED.h>.

I.e.
```
#include <SoftwareSerial.h>
#include <FastLED.h>
```
in your sketch/main.cpp.

If you don't include that there won't be `PIXIE` declared.
This should be it, not sure though

highoctane305, 2020-05-30:
Thanks for that, and sorry to be unclear - I've been including SoftwareSerial.h before FastLED.h for some years now and everything works well. It's ONLY because of changing to something newer than 2.5.2 of the ESP8266 software that the problem happens.

I can switch my board in the Arduino IDE to some AVR based Arduino (Uno, Mega etc) and exactly the same code will compile without fault, albeit changing a pin number where appropriate.

So there's some interesting tie up with FastLED and ESP8266 going on here.

5chmidti, 2020-05-31:
Yeah, sounded like you've used it before successfully.

Going down the rabbit hole  to [here](https://github.com/plerup/espsoftwareserial/blob/91ea6b1b1c34601565b23c96c4441f2d399a4f99/src/SoftwareSerial.h) showed that they changed the defined symbol name:
```
#ifndef __SoftwareSerial_h
#define __SoftwareSerial_h
```
I've opened a pull request #1032 that should fix it, it would be best if you could try that out and see if it fixes your problem ;)

highoctane305, 2020-06-01:
Wow! Yes that definitely did fix it for the ESP8266. Thank you.
There is still an issue with the ESP32 however. The same blink code (changing a pin number where appropriate), gets me this error when using espressif's ESP32 support however:

The code:
```
#include <SoftwareSerial.h>
#include <FastLED.h>

// How many leds in your strip?
#define NUM_LEDS 1
#define DATA_PIN 19

// Define the array of leds
CRGB leds[NUM_LEDS];

void setup() { 
    FastLED.addLeds<PIXIE, DATA_PIN, RGB>(leds, NUM_LEDS);  // GRB ordering is assumed
}

void loop() { 
  // Turn the LED on, then pause
  leds[0] = CRGB::Red;
  FastLED.show();
  delay(500);
  // Now turn the LED off, then pause
  leds[0] = CRGB::Black;
  FastLED.show();
  delay(500);
}
```

The error when compiling:
```
Arduino: 1.8.9 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

Build options changed, rebuilding all
In file included from /tmp/arduino_modified_sketch_756049/HelloWorld_i2c.pde:2:0:
/home/steven/Arduino/libraries/FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003
 #    pragma message "FastLED version 3.003.003"
                     ^
In file included from /home/steven/Arduino/libraries/FastLED/FastLED.h:65:0,
                 from /tmp/arduino_modified_sketch_756049/HelloWorld_i2c.pde:2:
/home/steven/Arduino/libraries/FastLED/fastspi.h:130:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
In file included from /home/steven/Arduino/libraries/FastLED/FastLED.h:66:0,
                 from /tmp/arduino_modified_sketch_756049/HelloWorld_i2c.pde:2:
/home/steven/Arduino/libraries/FastLED/chipsets.h: In member function 'virtual void PixieController<DATA_PIN, RGB_ORDER>::init()':
/home/steven/Arduino/libraries/FastLED/chipsets.h:36:22: error: no matching function for call to 'SoftwareSerial::begin(int)'
   Serial.begin(115200);
                      ^
In file included from /tmp/arduino_modified_sketch_756049/HelloWorld_i2c.pde:1:0:
/home/steven/Arduino/libraries/EspSoftwareSerial/src/SoftwareSerial.h:52:7: note: candidate: void SoftwareSerial::begin(int32_t, int8_t, int8_t, SoftwareSerialConfig, bool, int, int)
  void begin(int32_t baud, int8_t rxPin, int8_t txPin = -1,
       ^
/home/steven/Arduino/libraries/EspSoftwareSerial/src/SoftwareSerial.h:52:7: note:   candidate expects 7 arguments, 1 provided
In file included from /home/steven/Arduino/libraries/FastLED/FastLED.h:66:0,
                 from /tmp/arduino_modified_sketch_756049/HelloWorld_i2c.pde:2:
/home/steven/Arduino/libraries/FastLED/chipsets.h: In instantiation of 'PixieController<DATA_PIN, RGB_ORDER>::PixieController() [with unsigned char DATA_PIN = 19u; EOrder RGB_ORDER = (EOrder)10u]':
/home/steven/Arduino/libraries/FastLED/FastLED.h:88:52:   required from 'static CLEDController& CFastLED::addLeds(CRGB*, int, int) [with CHIPSET = PIXIE; unsigned char DATA_PIN = 19u; EOrder RGB_ORDER = (EOrder)10u]'
/tmp/arduino_modified_sketch_756049/HelloWorld_i2c.pde:14:57:   required from here
/home/steven/Arduino/libraries/FastLED/chipsets.h:32:41: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(int, unsigned char)'
  PixieController() : Serial(-1, DATA_PIN) {}
                                         ^
In file included from /tmp/arduino_modified_sketch_756049/HelloWorld_i2c.pde:1:0:
/home/steven/Arduino/libraries/EspSoftwareSerial/src/SoftwareSerial.h:48:2: note: candidate: SoftwareSerial::SoftwareSerial()
  SoftwareSerial();
  ^
/home/steven/Arduino/libraries/EspSoftwareSerial/src/SoftwareSerial.h:48:2: note:   candidate expects 0 arguments, 2 provided
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

highoctane305, 2020-06-02:
That last issue with the ESP32 goes away when I install the latest version (6.8.1 at the time of writing) of EspSoftwareSerial.
To me, this solves the problems I've been having. In my view, this issue can be closed.
Thank you.

kriegsman, 2020-07-29:
I believe this is now fixed as of https://github.com/FastLED/FastLED/pull/1032 
Thanks for the help!

closed 2020-07-29 by kriegsman