FastLED/FastLED #1034: Support for ESP32-S2

fabiuz7, 2020-06-06:
Hi,
I'm just testing the new [ESP32-S2 dev module](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-saola-1-v1.2.html). Is ESP32-s2 supported? Because the code compiles with [esp32s2 branch](https://github.com/espressif/arduino-esp32/tree/esp32s2), but the led doesn't turn on.

DrJohnM61, 2020-07-01:
I have tried the Adafruit Neopixel library and it works, however, it does not work if you include the WiFiMulti and issue a WiFiMulti.run() command. This may be related to https://www.eevblog.com/forum/beginners/esp12f-with-ws2812-(neopixels)-within-arduino/ and the suggestion is to use FastLED. Any suggestion of how to use FastLED on a ESP32S2 would be appreciated.

roblatour, 2020-07-23:
I have it part-way working .with this code and an espressif  esp32-s2-saola-1 board

`#include <FastLED.h>

#define LED_PIN     18
#define NUM_LEDS    1
#define BRIGHTNESS  10
#define LED_TYPE    WS2812
CRGB leds[NUM_LEDS];

enum colours { Green, Blue, Red };

void setNeoPixelColour(colours choosenColour) {

  // https://github.com/FastLED/FastLED/blob/master/examples/RGBSetDemo/RGBSetDemo.ino
  // https://github.com/FastLED/FastLED/wiki/FastLED-HSV-Colors

  switch (choosenColour) {
    case Green:
      Serial.println("green");
      leds[0] = CRGB::Green;
      break;
    case Blue:
      Serial.println("blue");
      leds[0]  = CRGB::Blue;
      break;
    case Red:
      Serial.println("red");
      leds[0] = CRGB::Red;
      break;
    default:
      // statements
      break;
  }

  FastLED.show();   // <<<< problem here
  
}
`

`FastLED.addLeds<WS2812, LED_PIN, RGB>(leds, NUM_LEDS);
  FastLED.setBrightness(BRIGHTNESS );`

`setNeoPixelColour(Green);`

It will turn on the led to the desired colour, but after the FastLED.show(); command is executed the board reboots.

Here is the dump:

Blue
E (7363) rmt: rmt_set_pin(480): RMT CHANNEL ERR
E (7363) rmt: rmt_config(607): set gpio for RMT driver failed
E (7363) rmt: rmt_set_tx_thr_intr_en(462): RMT CHANNEL ERR
E (7368) rmt: rmt_set_pin(480): RMT CHANNEL ERR
E (7373) rmt: rmt_config(607): set gpio for RMT driver failed
E (7378) rmt: rmt_set_tx_thr_intr_en(462): RMT CHANNEL ERR
E (7383) rmt: rmt_set_pin(480): RMT CHANNEL ERR
E (7388) rmt: rmt_config(607): set gpio for RMT driver failed
E (7393) rmt: rmt_set_tx_thr_intr_en(462): RMT CHANNEL ERR
E (7398) rmt: rmt_set_pin(480): RMT CHANNEL ERR
E (7403) rmt: rmt_config(607): set gpio for RMT driver failed
E (7408) rmt: rmt_set_tx_thr_intr_en(462): RMT CHANNEL ERR
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4002726b  PS      : 0x00050034  A0      : 0x400a0d02  A1      : 0x3ffbf950  
A2      : 0x00800000  A3      : 0x00838040  A4      : 0x28c08800  A5      : 0x3ffbf920  
A6      : 0x00000008  A7      : 0x00001000  A8      : 0x0000001f  A9      : 0x00000000  
A10     : 0x3ffc10f8  A11     : 0x00000001  A12     : 0x3ffc12a4  A13     : 0x3ffc1288  
A14     : 0x00000000  A15     : 0x00000015  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x3ffc12a4  LEND    : 0x3ffc1288  LCOUNT  : 0x40025d68  
Core  0 was running in ISR context:
EPC1    : 0x400a196b  EPC2    : 0x00000000  EPC3    : 0x400a0d02  EPC4    : 0x4002726b

Backtrace:0x40027268:0x3ffbf950 0x400a0cff:0x3ffc5290 0x4008e0cf:0x3ffc52b0 0x400299c1:0x3ffc52d0 0x400292a9:0x3ffc52f0

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025dd9
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x570
load:0x40050000,len:0x1620
load:0x40054000,len:0x1a04
entry 0x40050280

roblatour, 2020-07-23:
found this closed issue that seems to be related (perhaps because the esp32-s2 only has one core?):, 
https://github.com/FastLED/FastLED/issues/626

tried implementing a call to FastLEDshowESP32() as described here:
https://gist.github.com/hpwit/d29333138c7952d254ad90844802ee55
instead of FastLed.show() but both caused a reboot.

Clearly this issue is above my pay-grade.  Hope someone can help.

cmsavage, 2020-07-25:
@roblatour, could you try adding the following before importing FastLED.h:

`#define FASTLED_RMT_MAX_CHANNELS 1`

Looking through the code, the default behavior is to use 8 RMT channels, but the ESP32-S2 only has 4 RMT channels.  Looking at your dump, maybe FastLED is trying to initialize some RMT channels that do not exist?

roblatour, 2020-07-25:
Thank you but sadly
#define FASTLED_RMT_MAX_CHANNELS 1
did not help

roblatour, 2020-08-01:
I found a solution using the Adafruit Neopixel library ref: [https://github.com/adafruit/Adafruit_NeoPixel](url)
my code below:

`include <Adafruit_NeoPixel.h>

#define PIN         18   //Pin 18 is used on the ESP32-S2-SAOLA-1 board to address the NeoPixel
#define BRIGHTNESS  5

Adafruit_NeoPixel pixels(1, PIN, NEO_GRB + NEO_KHZ800);

enum colours { Off, Red, Green, Blue, Yellow };

void setNeoPixelColour(colours choosenColour) {

  byte red, green, blue;

  switch (choosenColour) {

    case Red:
      red = 255; green = 0; blue = 0;
      break;

    case Green:
      red = 0;  green = 255; blue = 0;
      break;

    case Blue:
      red = 0; green = 0; blue = 255;
      break;

    case Yellow:
      red = 255; green = 255; ; blue = 0;
      break;

    case Off:
      red = 0; green = 0; blue = 0;
      pixels.clear();
      break;

    default:
      break;
  }

  pixels.setPixelColor(0, red, green, blue);
  pixels.show();

}

//----------------------------------------------------------------

void setup()
{

  delay(3000);  //saftey delay at start-up

  pixels.begin();
  
  // the following statements prevent a bright green beacon from displaying at startup 
  pixels.setBrightness(0);
  pixels.show();
  setNeoPixelColour(Off);
  pixels.setBrightness(BRIGHTNESS);

}

void loop()
{

  setNeoPixelColour(Red);
  delay(5000);

  setNeoPixelColour(Green);
  delay(5000);

  setNeoPixelColour(Blue);
  delay(5000);

  setNeoPixelColour(Yellow);
  delay(5000);

  setNeoPixelColour(Off);
  delay(5000);

}`

rahmanshaber, 2021-03-24:
I tried the blink example with my esp32-s2 and it only stuck on whitesh color. Is the device not supported yet?

I tested the same code in the ardinuo micro and it works.

AshwiniNagelly, 2021-08-25:
@rahmanshaber have you solved the issue

9Rune5, 2021-11-28:
https://github.com/FastLED/FastLED/pull/1320 looks like it will address this issue

Tofapost, 2023-04-12:
Hey, So I had a similar issue, and the problem was in the libraries.txt file. 

`# Wokwi Library List
# See https://docs.wokwi.com/guides/libraries

FastLED@3.4.0`

Once I removed "@3.4.0" it worked. 

`# Wokwi Library List
# See https://docs.wokwi.com/guides/libraries

FastLED`

zackees, 2024-08-30:
Supported in idf 4.4, WS2812 will runtime assert in idf 5.x

closed 2026-04-17 by zackees