FastLED/FastLED #1035: NUM_LEDS

sanvd0000, 2020-06-06:
When using a simple code to run a led over 5 leds it keeps giving an error when NUM_LEDS is set to 5. It runs well when set to 6.

Now I kind of expect it has something to do with 0 being the first led, but still can't get my head around it. 

any ideas?

`// Use if you want to force the software SPI subsystem to be used for some reason (generally, you don't)
// #define FASTLED_FORCE_SOFTWARE_SPI
// Use if you want to force non-accelerated pin access (hint: you really don't, it breaks lots of things)
// #define FASTLED_FORCE_SOFTWARE_SPI
// #define FASTLED_FORCE_SOFTWARE_PINS
#include <FastLED.h>

// add const if this should never change
int interval = 100;
// Tracks the time since last event fired
unsigned long previousMillis = 0;

int whiteLed = 0;
int numled = 5;

///////////////////////////////////////////////////////////////////////////////////////////
//
// Move a white dot along the strip of leds.  This program simply shows how to configure the leds,
// and then how to turn a single pixel white and then off, moving down the line of pixels.
//

// How many leds are in the strip?
#define NUM_LEDS 5

// For led chips like WS2812, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806 define both DATA_PIN and CLOCK_PIN
// Clock pin only needed for SPI based chipsets when not using hardware SPI
#define DATA_PIN 23
//#define CLOCK_PIN 13

// This is an array of leds.  One item for each led in your strip.
CRGB leds[NUM_LEDS];

// This function sets up the ledsand tells the controller about them
void setup() {
  // sanity check delay - allows reprogramming if accidently blowing power w/leds
  delay(2000);

  Serial.begin(115200);

  FastLED.addLeds<WS2813, DATA_PIN, GRB>(leds, NUM_LEDS);

}

// This function runs over and over, and is where you do the magic to light
// your leds.
void loop() {
  unsigned long currentMillis = millis();

  if ((unsigned long)(currentMillis - previousMillis) >= interval) {
    Serial.println("check");
    Serial.println(whiteLed);

    if ( whiteLed < numled)
    {

      whiteLed++;

      leds[whiteLed] = CRGB::White;
      int oldwhiteLed = whiteLed - 1;

      leds[oldwhiteLed] = CRGB::Black;
      FastLED.show();

      Serial.println("New White");
      Serial.println(whiteLed);

    }

    if (whiteLed == numled) {
      Serial.println("Set white 0");
      whiteLed = 0;
    }
    if (whiteLed == 1) {

      leds[numled] = CRGB::Black;
      FastLED.show();
    }

    previousMillis = currentMillis;

    Serial.println("Extra check");
    Serial.println(whiteLed);

  }

}`

marmilicious, 2020-06-06:
@sanvd0000 When whiteled counts up and reaches 5 it will be trying to write pixel data to a pixel that doesn't exist.  This will cause bad things to happen in memory and all sorts of weird stuff can happen.  

We use github 'issues' here mostly just for bugs or improvements in the library itself.  For all general discussions, help, and how-to, please use the FastLED Reddit group here: https://www.reddit.com/r/FastLED

sanvd0000, 2020-06-06:
Hi @marmilicious thanks for you reply,  I understand but from my point of view there seems to be an error. 

When using the firstlight example from the library setting it to 60 leds, it will not light the last led in the string. 

Setting to 61 does the trick, but only 60 leds are actually in the string.
Using the led array going to 59 should sufficient but it is not the case.
Created this simple example to show. 

`// Use if you want to force the software SPI subsystem to be used for some reason (generally, you don't)
// #define FASTLED_FORCE_SOFTWARE_SPI
// Use if you want to force non-accelerated pin access (hint: you really don't, it breaks lots of things)
// #define FASTLED_FORCE_SOFTWARE_SPI
// #define FASTLED_FORCE_SOFTWARE_PINS
#include <FastLED.h>

///////////////////////////////////////////////////////////////////////////////////////////
//
// Move a white dot along the strip of leds.  This program simply shows how to configure the leds,
// and then how to turn a single pixel white and then off, moving down the line of pixels.
//

// How many leds are in the strip?
#define NUM_LEDS 60

// For led chips like WS2812, which have a data line, ground, and power, you just
// need to define DATA_PIN.  For led chipsets that are SPI based (four wires - data, clock,
// ground, and power), like the LPD8806 define both DATA_PIN and CLOCK_PIN
// Clock pin only needed for SPI based chipsets when not using hardware SPI
#define DATA_PIN 23
//#define CLOCK_PIN 13

// This is an array of leds.  One item for each led in your strip.
CRGB leds[NUM_LEDS];

// This function sets up the ledsand tells the controller about them
void setup() {
  // sanity check delay - allows reprogramming if accidently blowing power w/leds
  delay(2000);

  Serial.begin(115200);
  FastLED.addLeds<WS2813, DATA_PIN, GRB>(leds, NUM_LEDS);

}

// This function runs over and over, and is where you do the magic to light
// your leds.
void loop() {
  
// Move a single white led 
   for(int whiteLed = 0; whiteLed < NUM_LEDS;) {
      // Turn our current led on to white, then show the leds

      Serial.print("Turning On");
      Serial.println(whiteLed);

      leds[whiteLed] = CRGB::Green;
    
      // Show the leds (only one of which is set to white, from above)
      FastLED.show();

      // Wait a little bit
      delay(100);
      
      if(whiteLed - 1 >= 0){
      // Turn our current led back to black for the next loop around
      leds[whiteLed] = CRGB::Black; 
      }

      whiteLed++;
        
   }

}`

sanvd0000, 2020-06-06:
to me it seems that the code sees a led before leds[0], I just tested by setting the starting in to 1 and the result is exactly the same.

marmilicious, 2020-06-15:
You do not need that if statement inside your for loop.  Change the loop to something like this:

```
for (int whiteLed = 0; whiteLed < NUM_LEDS; whiteLed++) {
    Serial.print("whiteLed: "); Serial.println(whiteLed);  // help debug things
    leds[whiteLed] = CRGB::Green;  // set data to green
    FastLED.show();  // update the display
    delay(100);
    leds[whiteLed] = CRGB::Black; // set data to black
}
```

Please visit the FastLED reddit group for coding help.  https://www.reddit.com/r/FastLED  
Per the group rules there, when asking for help with your code please put it on pastebin.com or gist.github.com and share a **link** to your code.

closed 2024-10-12 by zackees