FastLED/FastLED #1037: One color of a ws2812 led strip shows NUM_LEDS further along a strip!?

fjalvingh, 2020-06-06:
I have 2 8x8 WS2812 led boards, one on top of another:

[ ] row 8..15
[ ] row 0..7

The bottom one's bottom row contains leds 0..7, the second row is 8..15 etc.

The leds are connected to an ESP32 board, and I wrote a trivial program (see below) to show each single color (R, G, B) for each row. So first row 0 gets red, then green, then blue; then row 1 gets red, then green, then blue... You get the picture. This program works fine on version 3.2.6 of the library.

But on later versions something odd happens: when the program is at row 0 it shows the green and blue colors, but it does not show the red. When it is supposed to show red the LEDs are off.

By lucky accident I connected two more of the 8x8 panels to the thing, so that the stack now looked like this:
[ ] row 24..31
[ ] row 16..23
[ ] row 8..15
[ ] row 0..7

Even though 256 LEDs are connected I leave NUM_LEDS at 128. I can now see where the red leds go: when the red leds for row 0 are supposed to show they are shown on row 16 instead(!). In fact, all red leds are shifted 16 rows (128 leds) UP.

It looks like somehow the NUM_LEDS parameter interferes with showing the RED color.

I have also tested with NUM_LEDS = 192, in that case the RED color for row 0 indeed shows up at row 24.

I tested the different versions of the library and the code last works correctly with version 3.2.6. After that:
3.2.7 does not compile,
3.2.8 makes all leds flash white
3.2.10 starts to exhibit the problem described here.

The program:

#include <FastLED.h>

#define LED_PIN     27
#define COLOR_ORDER BRG
#define CHIPSET     WS2812
#define NUM_LEDS    128

#define BRIGHTNESS  200
#define FRAMES_PER_SECOND 60

bool gReverseDirection = false;

CRGB leds[NUM_LEDS];

void setup() {
  delay(3000);
  
  FastLED.addLeds<CHIPSET, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS); //.setCorrection( TypicalLEDStrip );
  FastLED.setBrightness( BRIGHTNESS );
}

void setRowLeds(int row, CRGB color) {
  int ledix = 8*row;
  for(int i = 0; i < 8; i++) {
    leds[ledix++] = color;
  }
  FastLED.show();
}

void loop() {
  // In steps of 8 leds, cycle through all colors
  int row = 0;
  int maxrow = NUM_LEDS / 8;
  while(row < maxrow) {
    setRowLeds(row, CRGB::Red);
    delay(250);
    setRowLeds(row, CRGB::Green);
    delay(250);
    setRowLeds(row, CRGB::Blue);
    delay(250);
    setRowLeds(row, CRGB::Black);
    row++;
  }
}

SnowMB, 2020-07-10:
Hey,

Your problem might be the following:

Between setting a row to black and the next one to red (or between the last and the first row) you have no delay. The ESP32 is way faster to manipulate the LED data, than the actual data can be written to the leds. So you are basically calling `FastLED.show()` two times in a row. 

In the past I also have experienced problems with calling `FastLED.show()` to often with an esp32. For WS2812b it takes around 30µs (1.25µs per bit x 24 bit) per led to write the data. You can multiply this with you LED count to calculate what your maximum refresh rate for all of your leds is and how often you can call `FastLED.show()`.

The easiest fix would be to move `FastLED.show()` out of your `setRowLeds` function and just call it after `setRowLeds(row, CRGB::Red);`, `setRowLeds(row, CRGB::Green);` and `setRowLeds(row, CRGB::Blue);` respectively. But don't call it after `setRowLeds(row, CRGB::Black);`.

```c++
#include <FastLED.h>

#define LED_PIN 27
#define COLOR_ORDER BRG
#define CHIPSET WS2812
#define NUM_LEDS 128

#define BRIGHTNESS 200
#define FRAMES_PER_SECOND 60

bool gReverseDirection = false;

CRGB leds[NUM_LEDS];

void setup() {
delay(3000);

FastLED.addLeds<CHIPSET, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS); //.setCorrection( TypicalLEDStrip );
FastLED.setBrightness( BRIGHTNESS );
}

void setRowLeds(int row, CRGB color) {
int ledix = 8*row;
for(int i = 0; i < 8; i++) {
leds[ledix++] = color;
}
}

void loop() {
// In steps of 8 leds, cycle through all colors
int row = 0;
int maxrow = NUM_LEDS / 8;
while(row < maxrow) {
setRowLeds(row, CRGB::Red);
FastLED.show();
delay(250);
setRowLeds(row, CRGB::Green);
FastLED.show();
delay(250);
setRowLeds(row, CRGB::Blue);
FastLED.show();
delay(250);
setRowLeds(row, CRGB::Black);
row++;
}
}
```

I did a little digging and it seems like the CFastLED class has something like protection against calling `FastLED::show()` to often. 

https://github.com/FastLED/FastLED/blob/8b3877143287ac3e5d63934d1c8271c54035f243/FastLED.cpp#L45-L48

This uses a refresh value given by the controller class for the chipset and applied in `addLeds()`:

https://github.com/FastLED/FastLED/blob/8b3877143287ac3e5d63934d1c8271c54035f243/FastLED.cpp#L33-L43

However this does not seem to respect the number of leds, because for all of the esp32 controllers the maximum refresh rate is hard coded to 400.

https://github.com/FastLED/FastLED/blob/8b3877143287ac3e5d63934d1c8271c54035f243/platforms/esp/32/clockless_rmt_esp32.h#L246

https://github.com/FastLED/FastLED/blob/8b3877143287ac3e5d63934d1c8271c54035f243/platforms/esp/32/clockless_i2s_esp32.h#L233

https://github.com/FastLED/FastLED/blob/8b3877143287ac3e5d63934d1c8271c54035f243/platforms/esp/32/clockless_block_esp32.h#L71

But this can also be manually set via the `FastLED::setMaxRefreshRate` method. (But FastLED seems to block the show() method then which is not nice :thinking: ).

Maybe someone can clarify how this is supposed to work.

fjalvingh, 2020-07-21:
Hi @SnowMB, thanks a lot for your very detailed explanation and the time you spent to explain! I am currently not able to test this but I will do that as soon as I can (which will be around September). I will let you know the results.

zackees, 2026-08-20:
Resolved by #965, which added the required minimum low/latch interval between consecutive ESP32 RMT `show()` calls. The report was reproduced against 3.2.10; the fix is present from 3.4.0 onward and remains on current `master`.

The current ESP32 RMT implementation also encodes the chipset's RESET pulse at the end of every transmission, so a black frame followed immediately by the next color cannot be concatenated into one longer WS2812 frame. No user-added delay is required.

Closing as completed by the merged fix: https://github.com/FastLED/FastLED/pull/965

closed 2026-08-20 by zackees