FastLED/FastLED #1038: anyone tried HD107S?

sylvieuherkov, 2020-06-08:
APA102 LED has a high defective rate now, I am troubled with many defects. Someone sent me an email about HD107S, it says it can replace APA102, anyone tried it?

sylvieuherkov, 2020-06-28:
finally i bought HD107S strips sample from NEWSTAR LED https://www.addressableledstrip.com, hope the product is like as they said

ccoenen, 2020-07-03:
I have a few HD107S in use, they work like the APA102.

sylvieuherkov, 2020-07-13:
Really? but i am troubled with the high defective rate of APA102, how about the defective rate of HD107S?

ccoenen, 2020-07-13:
I didn't use either type in huge quentities, so far I have extremely few problems with the APA102 and none with the HD107S. But that's hardly statistically significant. (i.e. I hope someone else can answer this question).

sylvieuherkov, 2020-07-14:
thank you Claudius, then how many meters you used?  I am going to get one roll sample for testing, since our projects need a lot of APA102 stripes, if HD107S work well, it will solve a big problem for us.

ccoenen, 2020-07-14:
I soldered individual 5050 chips, no strips so far. The longest chains were probably only 10-15 leds long.

sylvieuherkov, 2020-07-15:
ok, I see, thank you, Claudius, our sample will be arriving these days, we'll see what's going on

closed 2026-08-20 by zackees