FastLED/FastLED #1040: ESP32 pin mask template breaking some SPI devices

fifthclassdesigns, 2020-06-15:
I've run into a strange issue where setting the clock pin of any SPI LED chipset to any GPIO < 32 crashes my SPI ethernet when calling FastLED.show(). Setting clock pin >= 32 or using clockless LEDs works fine.

I have a W5500 ethernet adapter on the VSPI bus, using Paul Stroffregen's standard Ethernet library. When I set the clock pin to any GPIO < 32, the link status of the W5500 phy goes haywire (false negatives and false positives). This happens regardless of what speed I set the SPI LEDs to, 1-24mhz. If I comment out FastLed.Show() in the main task I don't get the issue.

This issue occurs using the standard RMT driver, @hpwit's I2S driver, or @samguyer's brand new clockless driver with IRAM. Forcing software SPI doesn't fix the issue. 

[Example code to replicate this issue](https://pastebin.com/GdsJhg5r)

samguyer, 2020-06-15:
@fifthclassdesigns That's very strange. I'm not aware of any specific issues with pins less than 32. On the ESP32, you should be careful using pins 34-39 because they are designated by the hardware as input-only.

Can you give me a backtrace for the crash?

fifthclassdesigns, 2020-06-15:
@samguyer There's actually no crash/backtrace log, the ethernet phy just goes nuts when the APA clock pin is set to < 32 and show() is called. If I remove show() from my code, the ethernet behaves normally. 

The ethernet also works normally when clock pin = 32 or pin = 33, and when using clockless LEDs. Makes sense on 34-39, I'm not using them. I can post a scope trace if that's helpful?

samguyer, 2020-06-15:
Hmmm. When the situation occurs on different kinds of strips and different
drivers, my inclination is to look at what the *other* library is doing.

Another thing you could try is pushing FastLED show to the other core. Do
you know how to do that?

fifthclassdesigns, 2020-06-15:
Thanks for your help Sam--I've tried running FastLED show on both 0 and 1, and unfortunately there's no difference. The example code I linked above has ethernet handling on 0 and FastLED on 1. To clarify, the situation occurs no matter which FastLED driver I use, with any SPI LED chipset. 

Does it make sense that FastLED's different handling of bit masking for pins < 32 vs pins >= 32 could contribute to this? The bitwise operations happening in fastpin_esp32.h are above my pay grade, but given that modifying the one on line 32 to exclude the clock pin I'm using fixes the ethernet (albiet breaking the LEDs),  it seems related?

For reference, the ethernet library I'm using is the standard Arduino ethernet library (v2). The link status handling code in the library is pretty straightforward. It starts an SPI transaction, reads a register bit, ends the SPI transaction, and returns the phy status. 

At the moment, when clock pin < 32 and FastLED.show() is called, the link status flips back and forth between LINK_ON and LINK_OFF roughly every second because readPHYCFGR_W5500() returns 000000 once every 100ish calls when it shouldn't. That doesn't happen when I set clock pin to 32 or 33. Here's the relevant code from that library with my comments added:

```
SPI.beginTransaction(20000000, MSBFIRST, SPI_MODE0); // start SPI transaction at 20MHz, CPOL = 0 & CPHA = 0
phystatus = readPHYCFGR_W5500(); // read PHY Configuration register, default: 10111XXX. Returns 1000 00 when not connected, 1011 11 when connected
SPI.endTransaction(); // end SPI transaction
if (phystatus & 0x01) return LINK_ON; 
return LINK_OFF;
```

samguyer, 2020-06-15:
I'd recommend posting a question on the FastLED Subreddit to see if anyone else has input or advice. If it's a standard library, I'm guessing that others have used it successfully. Maybe there's some special trick to get them to work together.

hpwit, 2020-06-16:
Hello I have had a look at it couple of weeks ago and what bothers me is the TOGGLE function linked with the ‘clock’ when using ´spi’ on the esp32. Which is not really SPI ;) I will try to free sometime to look at it with the scope

fifthclassdesigns, 2020-06-16:
Thanks @samguyer. I posted there a few weeks ago and Yves mentioned that pins < 32 were handled differently than >= 32 on the ESP32, which is how I got started on this.

@hpwit, that makes sense given that modifying the if statement in the toggle function in the fastpin file to include the clock pin I'm using fixes the issue with the ethernet PHY (but breaks the LEDs). Let me know how I can help, happy to post scope traces of the line etc.

```
inline static void toggle() __attribute__ ((always_inline)) {
      if(PIN < 32 && PIN != 2) { GPIO.out ^= MASK; }
      else { GPIO.out1.val ^=MASK; }
  }
```

samguyer, 2020-06-16:
@fifthclassdesigns and @hpwit The toggle function could be an issue for clocked strips, but not for clockless strips. The RMT driver, for example, just loads up the buffer and the RMT peripheral does all the signal generation on the pins.

fifthclassdesigns, 2020-06-16:
@samguyer Ok, makes sense. This situation only occurs when using clocked strips, so that seems to track. 

I see that toggle is being used to avoid adjacent write opcodes to the same register, is there any reason why the timing would be handled differently internally by GPIO.out vs GPIO.out1?

fifthclassdesigns, 2020-06-16:
It seems the timing is different when calling FastLED.show() with clock on pin 2 vs pin 32/33

[image]

And it looks like when clock pin < 32 my ethernet CS pin is dropping data (?)

[image]

No issues on pin 32 or 33

[image]

fifthclassdesigns, 2020-06-18:
Ok, I reverted the relevant section of bitbang header file to [this code](https://github.com/FastLED/FastLED/commit/d7007f461fceabac086a8404ef7855032b3467ce) and the issue is fixed for me. 

@samguyer I realize this is a pretty niche issue, but is there any documentation you know of on FastLED's use of toggle to make sure the compiler isn't optimizing out two consecutive high calls to the same register? I see [Dan Garcia's commit](https://github.com/FastLED/FastLED/commit/5618f234a6c68b72413322f34d0f832e998b3ce0) adding it, but nothing else.

closed 2026-08-20 by zackees