FastLED/FastLED #1044: Red causes random flashing on diy arduino

achampagne1, 2020-06-24:
I created a DIY arduino using the attached schematic:
[image]
The Dout flag is the data line for the light strip, 5V of the strip is connected to the clean power flag, and ground of the strip is connected to ground. The whole circuit is being powered by usb from my laptop. I am driving 16 WS2812 LED (I know USB can not supply enough current for all at max brightness at once). I am using the code below to pulse the LED:
`
void pulse(int R,int G, int B){
  int brightness = 255;
  int fadeAmount = -1;
  for(int j=0;j<500;j++){
    for(int i = 0; i < NUM_LEDS; i++ ){
      leds[i].setRGB(R,G,B);
      leds[i].fadeLightBy(brightness);
    }
    FastLED.show();
    brightness = brightness + fadeAmount;
    if(brightness == 0 || brightness == 255){
      fadeAmount = -fadeAmount ; 
    }    
    delay(10);
  }
}
`
if I call the function with a tiny bit of red, like `pulse(10,110,0);` 
then the LED gently pulse, but a lot of the LED start flashing with random colors. This behavior does not happen if I call the function using just green `pulse(0,110,0);` or with just blue `pulse(0,0,110);`  Why would just the red color be causing this? I am using the arduino IDE to program the board.

SnowMB, 2020-07-10:
Formatted:

```c++
void pulse(int R,int G, int B)
{ 
  int brightness = 255; 
  int fadeAmount = -1; 
  for(int j=0;j<500;j++)
  { 
    for(int i = 0; i < NUM_LEDS; i++ )
    { 
      leds[i].setRGB(R,G,B); 
      leds[i].fadeLightBy(brightness); 
    } 
    FastLED.show(); 
    brightness = brightness + fadeAmount; 
    if(brightness == 0 || brightness == 255)
    { 
      fadeAmount = -fadeAmount ; 
    } 
    delay(10); 
  } 
}
```

Turbine1991, 2021-11-21:
I'm also getting this issue.

marmilicious, 2021-11-21:
You might want to post your question to the [FastLED reddit group](https://www.reddit.com/r/FastLED) where more people will see it.  Maybe you can share a video there too?  If you run this on a non-DIY (commercially available) UNO or similar do you see the flickering?

I tried your code (grabbed the formatted version that SnowMB posted) and also made this [modified version](https://pastebin.com/E7ArLqDA) and didn't see flickers (with my LPD8806 leds).  You didn't include a full program so I don't know what your master brightness was set to.  With a very low master brightness I do see stepping with your code, but not random flickering. 

It's best to not use delay() and let FastLED.show() run as often as possible to get the dithering.   https://github.com/FastLED/FastLED/wiki/FastLED-Temporal-Dithering

closed 2026-08-20 by zackees