FastLED/FastLED #1045: New Chip Type: NS108 / HD108

ccoenen, 2020-06-28:
Hi everyone, Newstar is producing a new chip with the same footprint as the APA102C/HD107/HD107s. The special thing about it is: it can do 16 bit per channel! It is also superior to the (already pretty great) APA102C in two other ways: it has 27KHz PWM and it undestands 3.0V as logic level "HIGH", so you could drive it at 5V and have 3.3V logic voltage without any tricks.

16bit is probably a major change to FastLED, but I think it would be amazing to support this chip.

A datasheet can be obtained here:
http://www.addressableledstrip.com/Datasheet-26.html (look for NS108/HD108)

The protocol is very straightforward and looks very much like the APA102-style protocols with the only change being the amount of bits sent:

* overall: 128bits `0` are the start frame
* 64 bits per LED:
  * 1 bit `1`
  * 15 bits gain adjustment (5 bit per channel, table in the datasheet)
  * 16 bits green
  * 16 bits red
  * 16 bits blue
* overall: 1bit `1` per led to end the frame

ccoenen, 2020-06-28:
As a quick and dirty first step, one could implement the protocol but left-shift the 8 bit data and just ignore or duplicate the least significant 8 bits. But eventual full 16 bit support would be amazing.

marmilicious, 2020-07-14:
Adding support for pixels with a higher number of bits per channel (and also support for RGBW) is definitely a future goal for FastLED.  And as you correctly guessed these changes will require a huge code update/rewrite due to how FastLED was originally optimized.  

Thank you for pointing out this new chip though and providing the details and datasheet link.  It does sound great!

sylvieuherkov, 2020-08-12:
> Adding support for pixels with a higher number of bits per channel (and also support for RGBW) is definitely a future goal for FastLED. And as you correctly guessed these changes will require a huge [...]

machinaut, 2020-11-08:
I just brought up a strip with these, and basically found the same.  In case its useful, here's another "brand" of the same LEDs, but the datasheet is worse/wrong in a bunch of places, which made bringup fun.  https://www.rose-lighting.com/wp-content/uploads/sites/53/2019/11/HD108-led.pdf

marmilicious, 2020-11-10:
Thank you @machinaut  
I'm going to link back to your reddit post about these too incase that's useful to anyone.  
https://www.reddit.com/r/FastLED/comments/jq646t/debugging_new_16bit_strands_hd108_rgb_leds/

machinaut, 2020-11-10:
@braydenm was able to get it in Adafruit_DotStar here as an example of integration (though it's pretty apparent how to update 8bit->16bit). https://github.com/braydenm/ledstrip

gee-ell, 2022-01-24:
thanks for you work guys, I've got 100 HD108 chips coming from China for a custom project.  Is it true (according to the buggy datasheet) that they don't require a decoupling capacitor?  And can someone confirm the pinout (designing a PCB)?

ccoenen, 2022-01-24:
I can't really tell you about the capacitors; i do not use any, but I also do not do product development with these (I only mess around with blinky stuff :-D ) and here it worked without caps nearby.

For the pinout: It is the exact same as APA102 or HD107. I am using them literally interchangeably with those (except for software, of course):

so:
```
              ____
 Data IN 1 --|    |-- 6 Data OUT
Clock IN 2 --|    |-- 5 Clock OUT
     GND 3 --|    |-- 4 VCC
              ¯¯¯¯
```

Here, the label still says APA102, because I am still using its footprint in KicadPCB.

[image]

gee-ell, 2022-01-24:
> I can't really tell you about the capacitors; i do not use any, but I also do not do product development with these (I only mess around with blinky stuff :-D ) and here it worked without caps [...]

Thanks @ccoenen.  I since found that the Newstar datasheet (which is appallingly bad - I mean do they actually want to sell their product??) _does_ mention capacitors, but you can't trust it, and the Rose Lighting one says you don't need them.  I've also seen pictures of HD107 strips that don't have any.  That's great but I wonder how they worked around that - from pictures of the chip it doesn't obviously seem to have an internal capacitor, but maybe it's directly integrated into the controller.

Shame HD108 hasn't caught on yet, 8bit is just awful the minute your dim and/or use gamma correction.

Has anybody found 2020's for sale at < 4000 MOQ yet?  I've not seen them for sale anywhere.

gee-ell, 2022-01-26:
@ccoenen, another Q for you.  I don't have the HD108 5050 chips yet, but my PCB layout for them relies on the exact location of the RGB diodes inside the chip.  A picture from an HD108 seller places them here:

[image]

as does the Newstar datasheet:

[image]

But the Rose Lighting datasheet shows them here:

[image]

which is correct?  or could those be a real layout difference between the Newstart and Rose Lighting versions?

ccoenen, 2022-01-26:
I got mine from aliexpress, so I couldn't even tell you who actually produced them. For previous types of LED (apa102/hd107..) I had different ones that were labelled the same. With that said: here's a photo of the ones I got a few months ago:
[image]

The three chips are "between pins 3 and 4" - so like in your upper graphic. Sorry for the bad photo, I just took a magnifying glass to my smartphone :D

ccoenen, 2022-01-26:
this is very similar to ones that were sold to me as APA102 (again: cheapest possible source, I have zero clue who made these or how I would get the exact same ones again :-D )
[image]

It's the exact same PCB layout as above, just with the older chip on it.

gee-ell, 2022-01-26:
Perfect thanks.  I got mine from AliExpress too (still in shipping).  Several sellers there, but I bet they're all the same company so most likely the same as yours.  Yours match the Newstar, down to the gold bond wires, so they're most likely Newstar.

jake653, 2022-01-27:
I have see, some ask "why 16 bit color"  it matters most if one is producing photographic images, in a video wall or spinning LED holographic fan.    There's noticeable dark scene banding when gamma correction is applied to 8 bit LED images.

 Examined 2 8-bit gamma tables, illustrating the loss of dark grays resolution:
    
```
                                 Unique outputs        bottom 128 out		bottom 64 out
    NewConvert.py        193				  38	                                11
    Adafruit Example     164                               54	                                 5 unique outputs
```
 
    Here's a Gamma article that may be of interest to some
    It references the 1993 paper: SMPTE93_Gamma.pdf
    I believe there was a human perception table published in the 1930s that may be a slightly  better correction guide than gamma.
   https://ledshield.wordpress.com/2012/11/13/led-brightness-to-your-eye-gamma-correction-no/
   
(One (not smart) LED Video Wall maker boasts having 72 bit color resolution for more natural images)

gee-ell, 2022-01-31:
> Perfect thanks. I got mine from AliExpress too (still in shipping). Several sellers there, but I bet they're all the same company so most likely the same as yours. Yours match the Newstar, down to the gold bond wires, so they're most likely Newstar.

So mine are the same as yours.  I asked the Ali seller if they were Newstar, and he said Newstar is actually just a distributor, and they both buy them from the same manufacturer.  So they are 'Newstar' in the sense that are the same as theirs & their datasheet (sorta) applies.

kasperkamperman, 2022-01-31:
An example on why 16bit color matters, a project I did a few years ago. 
https://www.youtube.com/watch?v=Ec1v1g_4LUI

This done with APA102 that has 5bit extra global brightness and dithering. Problem was that only the "original" APA102 has a linear 5 bit brightness. The SK clones (sold by many as APA102) have a different curve. 

Of course doing this with real 16 bit chips is easier.

gee-ell, 2022-02-01:
> Of course doing this with real 16 bit chips is easier.

Pretty.  Yeah slow fades are murder in 8bit.  I've actually hacked RGBWAUV stage lamps with my own microcontroller to do the same, for the same reasons (as well as making them wireless).  Some of those lamps are far worse than even 8bit, with massive steps in the dimmer levels.

So as I understand it, the 5 global bits are unique per LED?  Of course if you had a _really_ dim channel, and a really bright one, then you may have to clamp one to fit inside the global brightness dynamic range.  But HD108 would then give you way more than even 16bit (within the dynamic range limitation).

kasperkamperman, 2022-02-01:
Yes, the 5 bits are unique per led.

Kasper

Drizzt321, 2022-04-28:
This is awesome! Any further updates on getting the https://github.com/NaLG/FastLED_HD108/ upstreamed? Would be awesome to get this in so WLED and other projects can gain access to these new HD108s.

megazhuk, 2022-09-18:
Hi guys! Does anyone tries sk9826 led?
Can't get it works.. is just flickers at SPI data sends, and not latching. I think there are something wrong in pinout in datasheet or maybe was missed something else... When just CKI line is connected, without SDI on veery-low-speed data transfer (1kHz) and 1000ms delay between, led is turns-on on just CKI banging, and turns-off when delay. if i connect SDI line, then it turns-on more brighter and nothing else new. When change data speed to 1/2/4 MHz then it only flashing (data transfer shorter). 
Datasheet: [sk9826](https://ecksteinimg.de/Datasheet/LED/LED0032/SK9826_REV04_EN.pdf)
Photo:
[image]

megazhuk, 2022-09-19:
> Hi guys! Does anyone tries sk9826 led? Can't get it works.. is just flickers at SPI data sends, and not latching. I think there are something wrong in pinout in datasheet or maybe was missed something else... When just CKI line is connected, without SDI on veery-low-speed data transfer (1kHz) and 1000ms delay between, led is turns-on on just CKI banging, and turns-off when delay. if i connect SDI line, then it turns-on more brighter and nothing else new. When change data speed to 1/2/4 MHz then it only flashing (data transfer shorter). Datasheet: [sk9826](https://ecksteinimg.de/Datasheet/LED/LED0032/SK9826_REV04_EN.pdf) identical letter-by-letter to NS108E-5050 

I found it! There are wrong pinout in Datasheet.
[image]

On this chip pinout is:

May be it helps to someone.

cnheider, 2022-09-22:
Does anyone have a good small quantity supplier for HD108 strips? Please do share

Drizzt321, 2022-09-22:
@cnheider I actually found https://www.superlightingled.com/apa107hd107s-addressable-led-strips-c-5_487_354.html which has them, just make sure they're HD108, not the HD107. They group them together into the same product section. Seems to have reasonable prices for reasonable length of strips/IPx options.

boostednbagged, 2022-12-14:
> Does anyone have a good small quantity supplier for HD108 strips? Please do share

I placed a mass order to be able to get them at a decent price and control quality (super thick pcb's). What I didn't use I am reselling. I have 107 and 108 strips (both in 60 and 144).

I got the strips bc I wanted to use the 108s for my Ambilight set up but I could not get them to work with FastLED. So I am just following along, reading and learning lol.

zackees, 2022-12-14:
You don't need to change the RGB format in order to support these lights!

Hear me out - gamma correction!

I've literally been waiting for a decade for 16-bit LEDs like this so that gamma correction can be done without losing fidelity on the low end.

IMO the simplest path forward is to implement gamma correction in the driver so that FastLED still retains it's CRGB format as it's internal 24-bit color but then gamma converts to 48-bit color as it outputs the LED. The results would look fantastic - finally allowing dim colors to be implemented correctly. Almost every project I do I end up having to do gamma correction anyway so I would wonder if gamma correction could just be assumed for this strip type and manually disabled if someone didn't want it, which would be, effectively, never.

This is already what your computer screen does and people are fine with 24-bit color gamma corrected in hardware. The amount of code to support this would be minimal, where the alternative is just such a huge change.

Just my opinion, others may feel differently.

kasperkamperman, 2022-12-14:
Well, just converting it won't do. Because then you still have the big 8 bit steps. 
You need to add a smoothing algorithm that blends between two 8 bit values over time. 
```
interpolationFraction = interpolationFraction + interpolationStepsize;
if(interpolationFraction>254) interpolationFraction = 0;
```
You'll use this with a lerp function to blend between frame A and frame B.

zackees, 2022-12-14:
The stepping happens at mid and high ranges which is imperceptible.

But having a smoothing algorithm wouldn't be a bad thing - it's just complicated because the library is fed frame and not frames and timestamps to render asynchronously.

kasperkamperman, 2022-12-14:
`it's just complicated because the library is fed frame and not frames and timestamps to render asynchronously.`
Just render a new frame when the interpolationFraction = 0. Interpolate the rest with the previous frame.

zackees, 2022-12-14:
As far as I know there is no `interpolationFraction` currently implemented in fastled.

And like I said, I don't think you'll need it. Getting all 16 bits natively isn't going to deliver the increase in visual fidelity that you think it will. Professional graphics don't bother going much above 8-bit per color channel in the native format either because there is hardware gamma correction downstream from the frame buffer that makes it look awesome.

If it's good enough for PS4, your computer monitor and the 4k TV's out there, why wouldn't it be good enough for FastLED at this time?

kasperkamperman, 2022-12-14:
Well in order to do this product with very slow and smooth fades to black I needed interpolation. Just picking a 16bit gamma curve didn’t do the trick.

https://www.youtube.com/watch?v=Ec1v1g_4LUI

zackees, 2022-12-14:
This looks amazing. Can you show what it looks like when you just use a gamma function with no interpolation?

jake653, 2022-12-15:
Regarding color resolution..

I recall one video wall maker boasting of having 72 bit color resolution,
those LED panel bricks (getting down to 1 cent per pixel) rely on external
drivers to generate pwm dithering and multiplexing.

All that color resolution comes into the play in the darker part of the
color range, to reduce banding, etc. seen in darker scenes.

zackees, 2022-12-16:
Right, I'm sure it helps.

The issue is that the main developer of FastLED died in a freak fire accident and right now the main developer is mostly just mostly accepting pull requests from the community. A drastic change in the frame buffer format will touch a lot of code and this codebase is extremely hard to test for every platform. Actually it's worse than testing each platform, you also have to test each led type per platform using a real arduino. That's an MxN matrix that needs to be tested for this framebuffer change that's being proposed. If different settings are also tested then it's MxNxC.

That's just... intractable. 

A driver change to specific NS108/HD108 chipset doesn't really carry any of those risks, because the code is localized to just this led type and the gamma correction can be done in the MCU either as a function or a lookup table.

And that's why I'm pushing back by offering a simpler solution of on the fly gamma correction for just this LED type. Not because I believe my proposal better, but because it's something that can be implemented in a day with little to no risk to existing users.

kasperkamperman, 2022-12-20:
> This looks amazing. Can you show what it looks like when you just use a gamma function like with no interpolation?
> 
> ```c++
> uint16_t gamma(uint8_t x) } {
>   return uint16_t(x) * uint16_t(x);
> }
> ```
I'm not able to do that (most stuff is somewhere on the attic, because I moved). However, 256 steps stay 256 steps. Maybe you can improve distribution on the curve. 
This whole interpolation thing only matters when you do very slow fades to black (like I do over several seconds). Most people use FastLED for disco type effects, for which 8 bit is perfectly fine.

Joshfindit, 2022-12-20:
> Most people use FastLED for disco type effects, for which 8 bit is perfectly fine.

Personally I’m using it for room lamps, and am very interested in the smoothest most stable light output, especially at low brightness.

kasperkamperman, 2022-12-20:
FastLED has already built in dithering, so it provides a smooth curve on low light levels. 

However for slow fades from very bright to low brightness you need a chip with a higher range.

zackees, 2022-12-20:
Dithering is different from frame interpolation. The former requires no change to the frame buffer, the latter requires two frame buffers.

ccoenen, 2023-07-31:
There appears to be some movement on the forums: https://forum.electromage.com/t/preview-release-v3-45-new-hdr-led-drivers-faster-ws2812-and-editor-improvements/3045

zackees, 2025-04-08:
Follow up on this:

This chipset has 16 + 5 = 21 bits of brightness

This means we can do a full gamma correction curve on the uint8_t CRBG component.

We also have global brightness at 8bit. We can therefore get the full range of this LED.

banjostellar1-sketch, 2025-12-23:
Hello, I was hoping I could do some lightening for Christmas so i bought a bunch of the hd108 led. Turns out the fastled 3.10.3 doesn't officially support this led as the compiler returns an error "HD108 was not declared in this scope". It is clear fastled 3.10.4 is projected to add support for this led.  Question is; when will fastled 3.10.4 be released? was hoping I could use this led for my Christmas lightening project in few days time

nomakewan, 2025-12-24:
> Hello, I was hoping I could do some lightening for Christmas so i bought a bunch of the hd108 led. Turns out the fastled 3.10.3 doesn't officially support this led as the compiler returns an error [...]

https://www.reddit.com/r/FastLED/comments/1pu8avr/next_update_is_delayd_due_to_new_parlio_driver/

banjostellar1-sketch, 2025-12-26:
> https://www.reddit.com/r/FastLED/comments/1pu8avr/next_update_is_delayd_due_to_new_parlio_driver/

Thanks a lot

zackees, 2025-12-31:
https://github.com/FastLED/FastLED/pull/2119#issuecomment-3701506367

zackees, 2025-12-31:
So our HD108 protocal was just a copy of the apa102 algo and it was out of spec. But it worked for some but not all. This has been fixed in master and it should now run everywhere.