FastLED/FastLED #1046: Wrong output/flickering LED strip with Lolin ESP8266 NodeMCU V3, WS2812B

phlslv, 2020-06-28:
I am using a NodeMCU V3 with a WS2812B LED strip. I am also using a level shifter circuit.
When I try to run any of the examples for FastLED, the LED strip just doesn't light up correctly.
[There is a similar issue here with video.](https://github.com/FastLED/FastLED/issues/457)

[This is the code I am running.](https://pastebin.com/bijPtFKC) It is the DemoReel100 example with the pins modified. I also tried messing around with the FRAMES_PER_SECOND, but it's still flickering at the same rate...

I don't think the issue is with my hardware since the Neopixel examples run fine with my setup. Also I tried the FastLED example with an Arduino UNO and it also ran fine.

Any ideas on what could be causing this?

phlslv, 2020-07-03:
I tried setting BRIGHTNESS to 255 in the DemoReel100 example and the animation is much better. However there is still glitching expecially in the sinelon and juggle animations.

patni11, 2020-07-25:
havin similar issue, no idea how to solve but if I find something out I'll let you know

patni11, 2020-07-25:
Ok, so in my case. This worked. The problem were with the cables I used, I guess. But you can try this code

`//#define FASTLED_ESP8266_RAW_PIN_ORDER
//#define FASTLED_ESP8266_NODEMCU_PIN_ORDER
//#define FASTLED_INTERNAL
#include <FastLED.h>
#define LED_PIN     4
#define NUM_LEDS    50
CRGB leds[NUM_LEDS];

void setup() {
  FastLED.addLeds<WS2812B, LED_PIN, GRB>(leds, NUM_LEDS);
}

void loop(){
  for (int i = 0; i < NUM_LEDS; i++) {
    leds[i] = CRGB (0, 0, 255);
    FastLED.show(); 
    delay(40);
  }
  for (int i = NUM_LEDS-1; i >= 0; i --){
    leds[i] = CRGB (0, 255, 0);
    FastLED.show(); 
    delay(40);
  }
}

`

zackees, 2026-08-20:
Resolved by the ESP8266 clockless-output hardening in #1256 (merged 2021-07-18) and the subsequent interrupt-lock correction that remains in the current driver.

The merged fix detects an NMI/ISR-disturbed 24-bit pixel sequence, aborts it before it can be latched as corrupt data, waits the full reset interval, and retries the frame. It also raised the inter-frame wait to 85 µs for strips that require more than 70 µs. Current `clockless_esp8266.h` retains those protections and uses scoped interrupt locking so every early return unlocks correctly.

I also compiled the reporter's current path successfully:

```
bash compile esp8266 --examples DemoReel100 --defines FASTLED_ESP8266_RAW_PIN_ORDER --no-filter
```

Result: ESP8266 80 MHz build succeeded, including the raw GPIO mode used by the linked sketch. The other reproduction in this thread was ultimately traced to cabling, which is now covered by the general random-flash checklist added in #3931.

closed 2026-08-20 by zackees