FastLED/FastLED #1049: Having Fastled.Delay must after Show while using FastLed on ESP32 Amazon FreeRtos

pramodvb, 2020-07-10:
Hi, 
**My Environmen**t: ESP32 using FastLED along with AmazonFreeRTOS, connected with WS2812 LED Matrix of 8 LEDs x 8 LED
**Code**: FastLED.Show is called periodically in a High Priority task

**Observation without calling Fastled.Delay** after show, 
1. When I have 10 LEDs with Color Value and rest all with (0,0,0). In 1st  call with FastLED.show only 9 LEDs glow. In the subsequent calls to FastLED, 10 the LED also glows, but 3rd LED glows with altogether different color
2. When I send command to lit all 8 x 8 leds with constant color. The flicker is observed randomly.

**In order to address above:** 
1. I just added FastLED.Delay with 40 mSec right after call to .Show(). There is no flicker and there no Wierd color for 3rd LED also.

I thought of adding FastLED.delay after show after reading a few articles that If OS switches to another task while all data is not yet transferred completely. Considering FastLED.delay as blocking I

SnowMB, 2020-07-10:
Hey I just wrote about a similar issue in  #1037.

5chmidti, 2020-07-11:
`show()` and `delay()` are blocking, but there is one subtle difference in what `delay()` does.

`delay()` calls `show()` as well. It basically calls `show()` until the amount of milliseconds that were requested as a delay have passed.
But the notable thing about `delay()` is that it calls `yield()` after it's call to `show()`.

Try this:
```cpp
FastLED.show();
yield();
```

I'm not sure as to why `show()` doesn't just call `yield()` in the first place or why there aren't more people with this problem, if my assumptions are correct.

firepower9966, 2020-07-14:
What does yield do?

5chmidti, 2020-07-14:
[Here](https://stackoverflow.com/questions/34497758/what-is-the-secret-of-the-arduino-yieldfunction) is an explanation from stackoverflow.

I have to say though, that I'm not sure if it fixes the problem because I don't know what/if anything would be done in the background. I also don't know amazons FreeRtos, but the yield call is the only difference between `show()` and `delay()` in the FastLED functions, except for the delaying part.

closed 2026-08-20 by zackees