FastLED/FastLED #1051: Color correction on individual LED

tverspeelt, 2020-07-12:
Hi

I've built a string of APA102 LEDs in different sizes. I have 24 in 2020 format, 1 in 5050 format, and repeat that 6 times.

The 2020s have quite a bright red, I have to use a color correction value of 0xA0FFFF to get something close enough to white.
Problem is that the 5050s don't have this issue and also are a bit brighter than the 2020s.

Is there a method to apply a one time correction value to specific LEDs? FastLED.setCorrection does it global, something like FastLED.setCorrection[x] would be ideal in this situation.

If there is nothing built in, I guess I have no other option than adding a small correction routine before I run FastLED.Show.

Thanks

marmilicious, 2020-07-14:
Nothing on a per pixel basis that I know of.  As you mention, I think you'll have to run your own function to adjust things each time right before calling show().  

You might ask on the FastLED Reddit group https://www.reddit.com/r/FastLED to see if anyone has a creative idea.

tverspeelt, 2020-07-14:
The only downside I see from running my own correction before calling show(), is that the brightness of the 5050 LEDs will always be reduced.

With the global correction, I'm already turning down the red channel. But then I have to turn down green and blue even more to get the 5050 in the proper white range. I actually need the 5050s to be the brightest.
Maybe I could just set the global brightness higher, and then reduce the brightness on pixel level for the 2020s.

zackees, 2026-08-20:
Deferred during the bug burn-down: a persistent per-pixel correction API would require a public data-model/pipeline design and per-LED storage tradeoffs across memory-constrained targets. Existing code can apply correction to selected CRGB entries before show(); a library-level facility needs a separately prioritized design.