FastLED/FastLED #1052: Odd behavior using fill_rainbow() and Serial.begin()

FromenActual, 2020-07-14:
I've come across some weird edge case when using fill_rainbow(). This might actually not be an issue with FastLED, but this seems like the most logical place to post. Here's literally all of my Arduino code:

```
#include <FastLED.h>
#define NUM_LEDS 8
#define DATA_PIN 3
CRGB leds[NUM_LEDS];

void setup() { 
    //Serial.begin(9600);
    FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
    fill_rainbow(leds, NUM_LEDS, 60, 32);
    FastLED.show();
}

void loop() { 
}
```

As is, this works fine and makes a lovely rainbow. 

[image]

However if Serial.begin() is uncommented, the first LED goes red when it should be yellow. 

[image]

I first discovered this with a much more complicated sketch that included a hardware timer for another purpose, the code above is the simplest I could make and still recreate the bug.

In my testing, I found that an LED will only ever go red if its hue is 60. In some other tests with the serial monitor, I found the RGB value for that hue should be (160,150,0), however when the issue occurs the green component goes to 0. Those are the actual values being calculated, not just a wiring issue.

I looked at fill_rainbow() in colorutils.cpp, and noticed saturation is set to 240 instead of 255. Is there a reason for that? Regardless, changing saturation to 255 gets rid of this issue. To me that implies there could be some bug with adjusting for saturation in the HSV to RGB conversion, although I haven't looked too deeply there. If no bug is found there, could the default saturation for fill_rainbow() be set to 255 as a workaround?

Also, I've tested with both an Arduino Uno and Mega, which both exhibit the same issue. It's very odd that simply starting the serial communication causes a numerical calculation to be affected, perhaps there's some weird timing issue? I realize this is an extremely weird edge case and I don't personally mind whether this is addressed, but wanted to bring it up in case someone wants to take a stab at it.

marmilicious, 2020-07-15:
That is a weird one!
Curious what happens if you put a delay(3000) after the Serial.begin(9600) ?
Or what happens if you move Serial.begin(9600) to after FastLED.show() ?

FromenActual, 2020-07-15:
delay(3000) after Serial.begin(9600) -> LED still goes red

move Serial.begin(9600) to after FastLED.show() -> LED still goes red. Odd, I would have expected that to fix the issue.

Here's some other tests I ran, some of which are very odd:

Changing the baud rate appears to have no effect, LED always goes red.

Setting NUM_LEDS to 1-3 causes the LED to be yellow correctly. Setting to anything above 3 (only tested up to 8) causes the LED to turn red.

I set deltaHue to every value between -7 and +9, all of which make the first LED go yellow. Values lower than -7 and greater than +9 make it go red (at least, those that I've tested).

Changing the color ordering (RGB, GRB, etc.) has no effect on the calculation, it still calculates the first LED to be red (visually a different color, but still wrong)

Changing the LED type didn't change anything, first LED is red with WS2812B, WS2812, and WS2811 (don't know which other LED types use the same communication protocol)

That's all I've done so far today, it's very odd behavior. I don't understand how something like changing deltaHue affects the very first LED, when it should only affect the others. Have you (or anyone else reading) been able to replicate this? Do I just have a bad aura in my house?

5chmidti, 2020-07-15:
There is a pr open that sets the saturation to 255: #943 with a discussion about the underlying problem at #668. There seems to be some weird things going on with what the compiler does with the R1 register on avr.

FromenActual, 2020-07-15:
Ok, good to know this has already been made aware. Would you like to close this issue to remove clutter? Happy to run any other tests if you need me to :)

5chmidti, 2020-07-16:
Thats up to you, close it if you want to.

FromenActual, 2020-07-16:
Seems like this isn't an issue with FastLED and has already been made known, so I'll close this. Only thing I'd say is that the default saturation could be changed to 255 as a workaround, plus it makes the colors _slightly_ more vibrant (although it's not really noticeable, so I don't think it would hurt anyone). Just my 2 cents :)

closed 2020-07-16 by FromenActual