FastLED/FastLED #1055: Compile errors on Adafruit ItsyBitsy nRF52840 Express

savall21, 2020-07-19:
I'm getting compile errors in Arduino IDE when I try to compile FastLed examples for the Adafruit ItsyBitsy nRF52840 Express.  It compiled fine for a Arduino Uno and Nano 33 BLE.  I'm testing with FirstLight but I think its a compatibility problem with the board.  I read another issue and tried un-commenting two lines in fastpin_arm_nrf52.h:
#define NRF_P0_BASE   0x50000000UL
#define NRF_P1_BASE   0x50000300UL

Here are the compile errors:

Arduino: 1.8.13 (Windows 10), Board: "Adafruit ItsyBitsy nRF52840 Express, 0.2.11 SoftDevice s140 6.1.1, Level 0 (Release)"

In file included from C:\Users\savall21\Documents\Arduino\libraries\FastLED\examples\FirstLight\FirstLight.ino:6:

C:\Users\savall21\Documents\Arduino\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.002.010

   14 | #    pragma message "FastLED version 3.002.010"

      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~

In file included from C:\Users\savall21\Documents\Arduino\libraries\FastLED/FastLED.h:48,

                 from C:\Users\savall21\Documents\Arduino\libraries\FastLED\examples\FirstLight\FirstLight.ino:6:

C:\Users\savall21\Documents\Arduino\libraries\FastLED/fastpin.h: In instantiation of 'class FastPin<3>':

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:52:38:   required from 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback_InitializePinState() [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = RGB; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10]'

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:247:44:   required from 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback(uint16_t) [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = RGB; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10; uint16_t = short unsigned int]'

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:166:25:   required from 'void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::showPixels(PixelController<_RGB_ORDER>&) [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = RGB; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10]'

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:161:18:   required from here

C:\Users\savall21\Documents\Arduino\libraries\FastLED/fastpin.h:207:24: error: static assertion failed: Invalid pin specified

  207 |  static_assert(validpin(), "Invalid pin specified");

      |                ~~~~~~~~^~

In file included from C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/fastled_arm_nrf52.h:8,

                 from C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms.h:11,

                 from C:\Users\savall21\Documents\Arduino\libraries\FastLED/FastLED.h:52,

                 from C:\Users\savall21\Documents\Arduino\libraries\FastLED\examples\FirstLight\FirstLight.ino:6:

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h: In instantiation of 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback_InitializePwmInstance(NRF_PWM_Type*) [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = RGB; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10]':

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:248:47:   required from 'static void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::startPwmPlayback(uint16_t) [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = RGB; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10; uint16_t = short unsigned int]'

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:166:25:   required from 'void ClocklessController<_DATA_PIN, _T1, _T2, _T3, _RGB_ORDER, _XTRA0, _FLIP, _WAIT_TIME_MICROSECONDS>::showPixels(PixelController<_RGB_ORDER>&) [with unsigned char _DATA_PIN = 3; int _T1 = 6; int _T2 = 8; int _T3 = 6; EOrder _RGB_ORDER = RGB; int _XTRA0 = 0; bool _FLIP = false; int _WAIT_TIME_MICROSECONDS = 10]'

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:161:18:   required from here

C:\Users\savall21\Documents\Arduino\libraries\FastLED/platforms/arm/nrf52/clockless_arm_nrf52.h:62:55: error: 'nrf_pin' is not a member of 'FastPin<3>'

   62 |         pwm->PSEL.OUT[0] = FastPin<_DATA_PIN>::nrf_pin();

      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~

exit status 1

Error compiling for board Adafruit ItsyBitsy nRF52840 Express.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thanks in advance for the help.

sutaburosu, 2020-07-20:
FastLED support for nRF52 is quite new.  You're using v3.2.10 which is rather old.  Several nRF52 fixes have been incorporated since the v3.3.3 release.  Try using the master branch here;  I get some warnings, but at least compilation succeeds.  It would be useful if you could report whether the generated binary works for you.  I don't have such a device to test with.

henrygab, 2020-08-03:
@savall21 -- Every board needs to have its pinout defined.  It's not a trivial task to test, but **_for your own purposes_**, you could probably get it working fairly quickly.

The above suggest you are trying to use Arduino pin D3 ... but isn't that connected to a LED onboard?   If so, find an unused pin.

Start by reviewing [PR 1027](https://github.com/FastLED/FastLED/pull/1027/files) -- which just added the Circuit Playground Bluefruit.  (The ItsyBitsy nRF52840 is one board that I don't have.)

You will need to figure out which pins are usable on that board, and how the ARDUINO pins map to the nRF port/pin numbers (hint: look in variants subdirectory of the Adafruit hardware BSP files)..

Here's a few examples that (entirely untested and unsupported), may unblock you -- just put the following at the top of your sketch:
```C++
    _FL_DEFPIN( 8,  8, 0);    // P0.08 -- D8   /  DotStar Data
    _FL_DEFPIN( 6,  9, 1);    // P1.09 -- D6   /  DotStar Clock

    _FL_DEFPIN(14,  4, 0);    // P0.04 -- A0   /  D14
    _FL_DEFPIN(15, 30, 0);    // P0.30 -- A1   /  D15
    _FL_DEFPIN(16, 28, 0);    // P0.28 -- A2   /  D16
    _FL_DEFPIN(17, 31, 0);    // P0.31 -- A3   / D17
    _FL_DEFPIN(18,  2, 0);    // P0.02 -- A4   /  D18
    _FL_DEFPIN(19,  3, 0);    // P0.03 -- A5   /  D19
    _FL_DEFPIN(20,  5, 0);    // P0.05 -- A6   /  D10

    _FL_DEFPIN( 0, 25, 0);    // P0.25 -- D00 /  UART RX
    _FL_DEFPIN( 1, 24, 0);    // P0.24 -- D01 /  UART TX
    _FL_DEFPIN(21, 16, 0);    // P0.16 -- D21 /  SDA
    _FL_DEFPIN(22, 14, 0);    // P0.14 -- D22 /  SCL

    // D2, D5, D7, D9 .. D13 left as an exercise for the reader
```

Good luck!

henrygab, 2022-10-20:
If the board is uses FreeRTOS, then see:
* https://github.com/FastLED/FastLED/issues/747#issuecomment-1015098404
* https://github.com/FastLED/FastLED/issues/1055#issuecomment-667931207

If the board is using MBED OS, then see:
* https://github.com/FastLED/FastLED/issues/747#issuecomment-1015115120

The status has not changed since those were written.  Given the ubiquity of ESP32 solutions (e.g., [WLED](/aircoookie/wled) and [ESPixelStick](/forkineye/ESPixelStick)) I am unlikely to put significant additional effort into new support for nRF52.  This doesn't prevent anyone else from putting in the efforts to get this working under MBED OS ... it's open source, after all!

closed 2024-09-11 by zackees