FastLED/FastLED #1056: Control over 3000 WS2812B LEDs

daireod1, 2020-07-20:
Hello,

I am working on a project that requires 3780 WS2812B LEDs to be controlled. I am currently using an Arduino Mega 2560.

The plan is to use programme on processing that sends text over serial depending on what option is selected. This string looks like this "AAAAxxxBBBBBxxxCCCCxxx..." where the letters determine the colour each LED will light up. 

I have the strips all connected and powered up and they work successfully when using a simple fastled demo code.

They also worked using the programme described above for 540 LEDs and string characters being sent over serial.

However, when I try and increase the number of LEDs and number of characters in this code below it does not work as it uses 15428 bytes of dynamic memory when maximum is 8192.

Is there any alternatives that has enough dynamic memory to control this number of LEDs?

Another issue is when I lower this number to 1500, a number that falls below the dynamic memory limit. I send the 1500 characters over serial and nothing happens. It seems to only work when I sent <540 characters over serial.

Can someone help explain what the limitation here is?

`#include <FastLED.h>
#define NUM_LEDS 3780
// Data pin that led data will be written out over
#define DATA_PIN 5
#define BRIGHTNESS  32
CRGB leds[NUM_LEDS];
const int numChars = 3780;
char receivedChars[numChars];

boolean newData = false;

void setup() {
    Serial.begin(9600);
  //  Serial.println("<Arduino is ready>");
    FastLED.addLeds<WS2812B, DATA_PIN, GRB>(leds, NUM_LEDS);
    FastLED.setBrightness(BRIGHTNESS);

}

void loop() {
    recvWithStartEndMarkers();
    showNewData();
}

void recvWithStartEndMarkers() {
    static boolean recvInProgress = false;
    static int ndx = 0;
    char startMarker = '<';
    char endMarker = '>';
    char rc;
 
    while (Serial.available() > 0 && newData == false) {
        rc = Serial.read();

        if (recvInProgress == true) {
            if (rc != endMarker) {
                receivedChars[ndx] = rc;
                ndx++;
               if (ndx >= numChars) {
                  ndx = numChars - 1;
             }
            }
            else {
                receivedChars[ndx] = '\0'; // terminate the string
                recvInProgress = false;
                ndx = 0;
                newData = true;
            }
        }

        else if (rc == startMarker) {
            recvInProgress = true;
        }
    }
}

void showNewData() {
    if (newData == true) {
        Serial.println("This just in ... ");
        Serial.println(receivedChars);
        String   str = receivedChars;
     for(int whiteLed = 0; whiteLed < NUM_LEDS; whiteLed = whiteLed + 1) {
      // Turn our current led on to white, then show the leds
      if (str.charAt(whiteLed) == 'A') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'B') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'C') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'D') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'E') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'F') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'G') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'H') {
      leds[whiteLed] = CRGB::Green;
      }
      if (str.charAt(whiteLed) == 'z') {
      leds[whiteLed] = CRGB::Red;
      }
      } 
      // Show the leds (only one of which is set to white, from above)
      FastLED.show();

        newData = false;
    }
}`

sutaburosu, 2020-07-20:
This issue-tracker isn't the right place for help with your own code.  Please close this issue and follow up on your [r/FastLED thread](https://www.reddit.com/r/FastLED/comments/hulxqa/controlling_over_3000_leds/).

5chmidti, 2020-07-20:
`Serial.begin(9600);` sets your baud rate to 9600 bps, try setting it to 115200.
Although you only seem to transmit `NUM_LEDS` many characters, which should be within the bandwidth, if you try to achieve higher frame rates this will help. Whether it helps with the actual problem or not.