FastLED/FastLED #1062: ESP32.   setting 77 pixels on a controller messes up the next controller's output!?!?

mr-wiggle, 2020-08-02:
I've been struggling with this very obscure bug over the last few days.

To create the issue, set up two controllers for FastLED

>#define COLOR_ORDER RGB                                       
>#define LED_TYPE WS2812
>#define STRING_1_PIN 27
>#define STRING_2_PIN 26
>
>FastLED.addLeds<LED_TYPE, STRING_1_PIN, COLOR_ORDER>(leds, 0, string_1_num_leds);
>FastLED.addLeds<LED_TYPE, STRING_2_PIN, COLOR_ORDER>(leds, string_1_num_leds, string_2_num_leds);

If string_1_num_leds is set to 77 and string_2_num_leds is something higher than 77 (I'm using 100 for the example), the output of string 2 is corrupted.   

Here's the ouput of string 2, which is set for 100 pixels, on a logic analyzer,   When string 1 is set for 77 pixels, the output of string 2 cuts off after the first 8 bits of the 77th pixel.

[image]
Each pixel starts with a 1 followed by 23 zeros, so you can count the spikes as a pixel each.  Also, each 10th pixel sends out 8 1s then 16 0s, so they appear thicker in the image.

Here's a zoomed in view of the last couple pixels

[image]

If string 1 is set to 76,or 78 (or any other number besides 77 that I've teseted)  it works just fine.  I cannot see any reason why 77 should cause this issue.

If 3 strings are used and string 2 is set to 77 and string 3 is set to 100, the same problem happens on string 3.   3 will mess up 4.  4 will mess up 5 etc...

I don't know enough about the inner workings of the library to troubleshoot any further....

Here's a sketch that I wrote to test the issue.  It can run in Arduino on ESP32 with only the FastLED library installed.
The leds array is being set correctly.  You can see it print out on the serial monitor when the sketch is run.

[FastLED_77_pixel_bug.zip](https://github.com/FastLED/FastLED/files/5011520/FastLED_77_pixel_bug.zip)

> #include <FastLED.h>
>
>#define MAX_LEDS 400
>#define MAX_STRING_LEN 200
>#define COLOR_ORDER RGB                                       
>#define LED_TYPE WS2811    // WS2812
>#define STRING_1_PIN 27
>#define STRING_2_PIN 26
>
>uint32_t    string_1_num_leds = 77;                         // The number of LEDs in string 1
>uint32_t    string_2_num_leds = 100;                        // The number of LEDs in string 2
>uint32_t    NUM_LEDS = 0;                                   // total number of LEDs in the system
>CRGB        leds[MAX_LEDS];                                 // Initialize the LED array  
>
>
>/** Initialize FastLED */
>void initializeLEDstrings(){
>
>  // Check that no string exceeds MAX_STRING_LEN
>  if(string_1_num_leds < 0) string_1_num_leds = 0;
>  if(string_2_num_leds < 0) string_2_num_leds = 0;
>  if(string_1_num_leds > MAX_STRING_LEN) string_1_num_leds = MAX_STRING_LEN;
>  if(string_2_num_leds > MAX_STRING_LEN) string_2_num_leds = MAX_STRING_LEN;
>
>  // initialize the LED strings
>  FastLED.addLeds<LED_TYPE, STRING_1_PIN, COLOR_ORDER>(leds, 0, string_1_num_leds);
>  FastLED.addLeds<LED_TYPE, STRING_2_PIN, COLOR_ORDER>(leds, string_1_num_leds, string_2_num_leds);
>
>  // set the total number of LEDs
>  NUM_LEDS = string_1_num_leds + string_2_num_leds; 
>
>  fill_solid(leds, MAX_LEDS, CRGB::Black);  // start off with an array of black.
>
>}
>
>/**  get a hex color string from a CRGB color */
>String get_hex_color_from_crgb( CRGB color ){
>  long HexRGB = ((long)color.r << 16) | ((long)color.g << 8 ) | (long)color.b; // get value and convert.
>  return padLeadingZeros(String(HexRGB, HEX), 6, true);
>}
>
>/** Pad leading zeros or spaces for a string  */
>String padLeadingZeros(String str, int newStrLen, bool padWithZeros){
>
>  String padding = " ";
>  if(padWithZeros){ padding = "0"; }
>
>  int str_length = str.length();
>
>  if(newStrLen <= str_length) { return str; }
>  
>  for (int i = 0; i < newStrLen - str_length; i++){
>    str = padding + str;
>  }
>  
>  return str;
>}
>
>
>/** Log the LED array to the serial monitor */
>void logLedArray(){
>  int column = 10;
>  int row = 0;
>  Serial.print("\n\n");
>  for (int i=0; i < MAX_LEDS; i++){
>    if(column >= 10){
>      column = 0;
>      row += 10;
>      Serial.print("\n");
>      Serial.print( padLeadingZeros( String(row), 6, false)  );
>      Serial.print("   ");
>    }
>    column++;
>    Serial.print(get_hex_color_from_crgb( leds[i] ).c_str());
>    Serial.print("  ");
>  }  
>  Serial.print("\n");
>}
>
>
>void setup() {
>
>  Serial.begin(115200);
>
>  initializeLEDstrings();
>
>  log_i("Controller 0 size: %d",FastLED[0].size());
>  log_i("Controller 1 size: %d",FastLED[1].size());
>
>  fill_solid(leds, MAX_LEDS, CRGB(0x800000));  // 10000000,00000000,00000000 in binary
>
>  // set every 10th bulb to Red for easy counting
>  for (int i = 9; i< MAX_LEDS; i+=10){
>    leds[i] = CRGB(0xFF0000);
>  }
>  
>  logLedArray();
>
>}
>
>void loop() {
>  
>  FastLED.show(); 
>
>  delay(200);  //update bulbs every 200 ms
>  
>}
 
Thanks in advance for any help!

VinceW31, 2020-10-04:
Replace FastLED.show() with FastLED.delay(1) and see what happens

mr-wiggle, 2020-10-04:
No change in behaviour with FastLED.delay(1) instead of FastLED.show()

Also I have found that the same issue happens when string one is set to 21 bulbs.  So  21 and 77 both do it.

hazcat, 2020-12-01:
This may be due to a bug I've found in clockless_rmt_esp32.cpp

In ESP32RMTController::fillNext change the else block to read
`        } else {
            // -- No more data; signal to the RMT we are done
            for (uint32_t j = 0; j < 32; j++) {
                *pItem++ = 0; // this used to be writing to mRMT_mem_ptr
            }
        }`

writing to mRMT_mem_ptr stomps over values that were just written

mr-wiggle, 2020-12-01:
Hi hazcat,

  Thanks for the insight.   I'm running release version 3.3.3, which is structured a bit differently than the most current on here.

How would you apply the same fix to the code in release 3.3.3?    Here's the fillNext() function from clockless_rmt_esp32.h from that release:

```
void IRAM_ATTR fillNext() 
  {
        if (mPixels->has(1)) {
            uint32_t t1 = __clock_cycles();

            uint32_t one_val = mOne.val;
            uint32_t zero_val = mZero.val;

            // -- Get a pixel's worth of data
            uint8_t byte0 = mPixels->loadAndScale0();
            uint8_t byte1 = mPixels->loadAndScale1();
            uint8_t byte2 = mPixels->loadAndScale2();
            mPixels->advanceData();
            mPixels->stepDithering();

            // -- Fill 24 slots in the RMT memory
            register uint32_t pixel = byte0 << 24 | byte1 << 16 | byte2 << 8;

            // -- Use locals for speed
            volatile register uint32_t * pItem =  mRMT_mem_ptr;
            register uint16_t curPulse = mCurPulse;

            // Shift bits out, MSB first, setting RMTMEM.chan[n].data32[x] to the
            // rmt_item32_t value corresponding to the buffered bit value
            for (register uint32_t j = 0; j < 24; j++) {
                uint32_t val = (pixel & 0x80000000L) ? one_val : zero_val;
                *pItem++ = val;
                // Replaces: RMTMEM.chan[mRMT_channel].data32[mCurPulse].val = val;

                pixel <<= 1;
                curPulse++;

                if (curPulse == MAX_PULSES) {
                    pItem = & (RMTMEM.chan[mRMT_channel].data32[0].val);
                    curPulse = 0;
                }
            }

            // -- Store the new values back into the object
            mCurPulse = curPulse;
            mRMT_mem_ptr = pItem;
        } else {
            // -- No more data; signal to the RMT we are done
            for (uint32_t j = 0; j < 8; j++) {
                * mRMT_mem_ptr++ = 0;
            }
        }
    }
```

hazcat, 2020-12-01:
It looks like the issue I described doesn't exist in the 3.3.3 release. Sorry for any false hope I added ☹️

samguyer, 2020-12-30:
@hazcat Can I close this issue?

hazcat, 2020-12-30:
> @hazcat Can I close this issue?

@mr-wiggle should be the one to see if the issue is fixed.

I posted on this issue because I thought it might be a duplicate of a bug that I found.

hazcat, 2020-12-31:
I believe this issue is fixed.

I reproduced the issue @mr-wiggle described on the 3.3.3 release of the library 
https://youtu.be/tyKgunPg888

and it doesn't repro when using the 3.4.0 release. 
https://youtu.be/yalmrh-Z9LE

samguyer, 2020-12-31:
@hazcat Great! Thank you for testing it out. I'm going to close this issue. @mr-wiggle can reopen it if the problem is not fixed for him.

closed 2020-12-31 by samguyer

mr-wiggle, 2020-12-31:
Thanks for the support.  Confirmed fixed for me in 3.4.0 release