FastLED/FastLED #1064: Apa102 2020 LEDS appearing as dots

PhilColbert, 2020-08-05:
I am just using the fastled library to create POV images.

Using the 5050 apa102s the images come out clear and solid.

I moved now to 2020 's and the led seems to be firing and then going black creating a dot effect instead of the led staying on before changing to the new colour.

Do I need to change data_rate_mhz or ?

Anyone any ideas?

Thanks

marmilicious, 2020-08-06:
I'm assuming you're running the exact same code here?  
You can try lowering the data rate like this (example here setting it to 4 MHz, down from default of 12 MHz):
`
FastLED.addLeds<APA102, DATA_PIN, CLOCK_PIN, RGB, DATA_RATE_MHZ(4)>(leds,NUM_LEDS);
`

Are you sure they're APA102 chips?  Could they be SK9822?  
What controller are you using?  If it's a 3.3V output, are you using a level shifter?

PhilColbert, 2020-08-07:
Thanks for the reply, have tried different speeds.

Was sold as apa102's

esp32 and 3.3v output - maybe level shifter might work! 

thanks