FastLED/FastLED #1065: Need to increase # of leds per strip using fastled on a teensy 4

TreeDan, 2020-08-07:
I have 8 50-led ws2811 strings connected for a total of 400 leds.  I was using an arduino due with fastled and was able to do a fastled.show() to all 400 with no problem.  I just purchased a teensy 4.1 and used the code shown below written by Chemdoc77. Sketch works fine as long as i keep NUM_LEDS_PER_STRIP below 71. Does anybody know how i can increase to 400!

Sketch:
/* New Product Test Sketch by Chemdoc77
   used to test new RGB LED strips and Matrix that I purchase.
   Note:  Uses FastLED's Parallel Output procedure to work with a Teensy 4.0
*/
#include <FastLED.h>
#define DATA_PIN  7  // it is critical that you use this pin number or other pin numbers using FastLED's Parallel Output procedure
int brightness = 40;

// Teensy 4.0 Stuff ================
//#define NUM_LEDS_PER_STRIP 64
#define NUM_LEDS_PER_STRIP 71
#define NUM_STRIPS 1 
#define NUM_LEDS   NUM_LEDS_PER_STRIP  
CRGB leds[NUM_LEDS_PER_STRIP * NUM_STRIPS];

void setup() { 
  delay(1000);
  // Teensy 4.0 Stuff ==============
//   FastLED.addLeds<NUM_STRIPS, WS2812B,DATA_PIN,GRB>(leds, NUM_LEDS_PER_STRIP);
  //USE for WS2811:      
  FastLED.addLeds<NUM_STRIPS, WS2811, DATA_PIN >(leds, NUM_LEDS_PER_STRIP);
  FastLED.setBrightness(brightness);
//  FastLED.setMaxPowerInVoltsAndMilliamps(5, 1500);
//   set_max_power_indicator_LED(13);
  fill_solid(leds, NUM_LEDS, CRGB::Black);
  FastLED.show();
}
void loop() {  
  delay(3000);
       cd77_colorwipe_dot(CRGB::Red, 0 , NUM_LEDS, 20);
}
//==================== Functions ===============================
void cd77_colorwipe_dot(CRGB color, uint16_t from,  uint16_t to,  uint16_t wait) {
  for (uint16_t i = from; i <to; i++) {
    leds[i] = color;
    FastLED.delay(wait);
    leds[i] = CRGB::Black;
    FastLED.show();
  }
}

dpratt4, 2021-07-11:
I had the same problem on Teensy 4.0.  Turning down the refresh rate did the trick for me.
FastLED.setMaxRefreshRate(FramesPerSecond)

kcouck, 2021-07-11:
Hi, if you have 8 strips of 50 LEDs, NUM_LEDS_PER_STRIP = 50 should be enough?
If you put 400 you're trying to allocate memory for 3200 LED's...

Additionally, I've lost a year and a half trying to figure out why newer LED strips were not working like my old ones, and the issue seemed to be that there was not enough time between sending the last byte and the start of the new command sequence (hence never a long 'LOW' and no 'reset')
FastLED.delay() causes this problem as well as it doesn't really delay, it just repeats the input until the delay passes.
Driving 16 strips with NUM_LEDS_PER_STRIP=136 here with a Teensy 3.6, so a 4.1 should be able to handle that fine :)

I removed the .delay and put a maxrefreshrate(100) as dpratt4 suggested and the issue got solved. (I'm trying to really push a different frame at 100fps :p) (I might have upped the reset length as well in my FastLED copy, I don't completely remember)