FastLED/FastLED #1067: Compiler warning (unused variable) in /platforms/esp/32/clockless_rmt_esp32.h

bdaroz, 2020-08-09:
From FastLED 3.3.3 on ESP32.

Compiler Warning (truncated and redacted):

```
Users/xxxxxxx/Documents/Arduino/libraries/FastLED/platforms/esp/32/clockless_rmt_esp32.h:511:22: warning: unused variable 't1' [-Wunused-variable]
             uint32_t t1 = __clock_cycles();
```

kriegsman, 2020-09-04:
Thank you for the report, @bdaroz.  
@samguyer could you take a look and see if this is easy to fix?

samguyer, 2020-09-04:
I'll take a look. But I've also got some bigger changes to push, so maybe it's better to wait.

Account-Git, 2020-12-17:
I just wanted to bump this issue because it has been over 3 months now and it is still giving compiler warnings.

samguyer, 2020-12-17:
@Account-Git 
I don't think this warning exists in the current head.

closed 2020-12-30 by samguyer