FastLED/FastLED #1068: RGBW Support

MLDMoritz, 2020-08-10:
Hey,

as it seems there is still no RGBW support - How can I help?

tmanaud, 2020-09-10:
AFAIK there is no native RGBW but I daily use FastLED with SK6812 RGBW strips with this hack:  
https://www.partsnotincluded.com/fastled-rgbw-neopixels-sk6812/

Alexandr180, 2020-11-28:
A very important topic for me! Not clear if all effects work with this solution? Especially Fire2012 \ 18? noise effect and others?

tmanaud, 2020-11-28:
> A very important topic for me! Not clear if all effects work with this solution? Especially Fire2012 \ 18? noise effect and others?

Yes, besides performances issue with a cheap Arduino a more than 500 leds, everything is good.

leonbohmann, 2021-01-07:
And there still is an open issue in here: https://github.com/FastLED/FastLED/issues/106

@focalintent Are there any updates regarding this?

marmilicious, 2021-01-09:
@leonbohmann  Sorry to inform you, but Dan (focalintent) isn't with us any more.  
https://www.reddit.com/r/FastLED/comments/czenwv/sad_news/

And no, FastLED does not have RGBW support yet.

leonbohmann, 2021-01-09:
Oh man, that's awful. Have been using FastLED for years now. 

If any help is needed in implementing the RGBW features I'd be glad to help out somewhere sometime.

Nears, 2021-05-15:
Is there any indication when this feature may be available? E.g. is it planned for the next release?

sofakng, 2021-12-04:
I'm also hoping for RGBW support ... (specifically SK6812 RGBW)

Gavin-Perry, 2022-01-03:
Also looking for RGBW support WS2814 etc

ModischFabrications, 2022-12-30:
[Suppliers](https://www.aliexpress.com/item/32476317187.html) have since caught up and offer the SK6812 for a resonable price, I would also like to see support for RGBW in FastLED to use them. https://github.com/FastLED/FastLED/wiki/Chipset-reference references some RGBW in the works, but I was unable to find a reference to it. Did anyone find an alternative solution yet?

zackees, 2022-12-30:
My suggestion is to implement an RGBW algorithm yourself in FastLED and upstream it with a pull request to mainline.

My humble suggestion for an RGBW algorithm would be to translate RGB values into RGBW values on the fly. Exposing the 4th component (W) to the developer is problematic because changing the framebuffer format from CRGB to CRGBW would be a **deep** and risky change since FastLED is designed to run on a matrix of different chipsets with different configurations.

However, the translation of RGB -> RGBW is tricky and should be done by someone with a deep understanding of light mixing to make it look better than just using the RGB components. Either as a direct mapping of RGB -> RGBW, or RGB -> HSV -> RGBW.

I went ahead and asked chat GPT how this could be done and it gave this response:

```C++
#include <iostream>

struct RGB {
  int r;
  int g;
  int b;
};

struct RGBW {
  int r;
  int g;
  int b;
  int w;
};

RGBW convertRGBtoRGBW(RGB rgb) {
  RGBW rgbw;
  rgbw.r = rgb.r;
  rgbw.g = rgb.g;
  rgbw.b = rgb.b;
  rgbw.w = std::max(0, std::min(255, 255 - (rgb.r + rgb.g + rgb.b) / 3));
  return rgbw;
}

int main() {
  RGB rgb{255, 0, 0};
  RGBW rgbw = convertRGBtoRGBW(rgb);
  std::cout << "R: " << rgbw.r << ", G: " << rgbw.g << ", B: " << rgbw.b << ", W: " << rgbw.w << std::endl;
  return 0;
}
/*
This function takes in an RGB struct as an argument and returns an RGBW struct. The RGBW struct has four
fields: r, g, b, and w, corresponding to the red, green, blue, and white channels, respectively. The conversion
process is simple: the values of the r, g, and b fields are simply copied over from the input RGB struct. The
value of the w field is calculated as the maximum of 0 and the minimum of 255 and 255 minus the average of
the r, g, and b fields. This ensures that the value of the w field is always between 0 and 255, inclusive.
*/
```

ModischFabrications, 2022-12-31:
I'm not sure I understand you correctly. Is the "wire" protocol for these new chips already supported? I could find any references in the wiki. I don't see how the RGBW translation can be used if the module can't be instructed to display any colors, regardless if RGB or RGBW.

zackees, 2022-12-31:
There's been a push to change the frame buffer format on FastLed for 16 bit and rgbw.

I wanted to get ahead of thay.

No i don't think there is a wire formate yet for RGBW.

zackees, 2023-01-01:
But if you wanted an implementation and have some spar RGBW to send me I will implement a device driver for this led.

ModischFabrications, 2023-01-05:
I haven't ordered any yet, I'm still in the early concept stages for my project. The link above would be one of my picks to order a few, prices are cheap and the seller is recommendable. 
Is there a "development fund" e.g. from donations or a bounty board, where you could request one to be ordered to you? I'm just a user, not involved in any way. If there is none 
I can offer to donate/gift some "for the cause", but a regular process with contribution tracking is probably easier. Can someone from the development team chime in with some background?

kriegsman, 2023-09-13:
Cross-referencing this here:  I've (finally) started work on the low-level FastLED driver for four-channel (RGBW) WS281x LED strips.  I'll be tracking that work over on this Issue:  https://github.com/FastLED/FastLED/issues/1540

zackees, 2024-10-12:
Implemented at the driver level. ESP32 has native support. The rest of the boards can use the emulated driver that spoofs RGBW data as RGB data + padding and sends it down to the driver.

closed 2024-10-12 by zackees