FastLED/FastLED #1069: Custom colours in Pacifica example

ChopsKingsland, 2020-08-14:
I'm trying to customise the colours that are used in the Pacifica (https://github.com/FastLED/FastLED/blob/master/examples/Pacifica/Pacifica.ino) example, but I can't work out how to do it... I googled around and got to this website: http://fastled.io/docs/3.1/class_c_r_g_b_palette16.html, but that doesn't help me at all... Can I define the colours using RGB?

5chmidti, 2020-08-14:
```c++
CRGBPalette16 pacifica_palette_1 = 
    { 0x000507, 0x000409, 0x00030B, 0x00030D, 0x000210, 0x000212, 0x000114, 0x000117, 
      0x000019, 0x00001C, 0x000026, 0x000031, 0x00003B, 0x000046, 0x14554B, 0x28AA50 };
```
constructs a `CRGBPalette16` from 16 colors (in your linked docs the first constructor/member function).

That specific constructor takes 16 `CRGB` objects as arguments.
These `CRGB` objects are implicitly converted to from the hex values like `0x000507` via this constructor of `CRGB`:
```c++
    /// allow construction from 32-bit (really 24-bit) bit 0xRRGGBB color code
    inline CRGB( uint32_t colorcode)  __attribute__((always_inline))
    : r((colorcode >> 16) & 0xFF), g((colorcode >> 8) & 0xFF), b((colorcode >> 0) & 0xFF)
    {
    }
```
If you want to do it this way, google for hex colors.

But `CRGB` objects can also be created via other constructors (click on `CRGB` in the docs).
What you may be looking for is
```c++
    /// allow construction from R, G, B
    inline CRGB( uint8_t ir, uint8_t ig, uint8_t ib)  __attribute__((always_inline))
        : r(ir), g(ig), b(ib)
    {
    }
```

To use this constructor to create the temporary colors to create the palette, write for example
```c++
CRGBPalette16 pacifica_palette_1 = 
    { CRGB(255,12,123), /*the 15 other colors...*/ };
```

More on colors (as a write up rather than api docs) [here](https://github.com/FastLED/FastLED/wiki/Pixel-reference) in the [github wiki](https://github.com/FastLED/FastLED/wiki).

Note: if you have more questions please ask them in the [subreddit](https://www.reddit.com/r/FastLed)

kriegsman, 2020-09-04:
Thank you, @5chmidti -- great explanation.  Thank you for helping out!

closed 2020-09-04 by kriegsman