FastLED/FastLED #1070: Multiple LED arrays, updating each array individually

QuitButton, 2020-08-16:
I have two LED arrays, for example...

```
CRGB LEDs[NUM_LEDS];
CRGB COLONs[2];
FastLED.addLeds<WS2812B, DATA_PIN, GRB>(LEDs, NUM_LEDS); // clock display
FastLED.addLeds<WS2812B, COLON_PIN, GRB>(COLONs,2); // 2 leds for centre colon.
```

calling FastLED.show(); updates all leds, but I want to just update the 2 LEDs on in the COLON array. Is this possible? The Multiple-Controller-Examples only seem to show examples using the same number of leds per strip.

5chmidti, 2020-08-16:
I belive it's
```c++
FastLED[1].showLeds(/*optional brightness uint8_t*/);
```
to update your second strip.

marmilicious, 2020-08-16:
@QuitButton  Check out the last section of this wiki page.  
https://github.com/FastLED/FastLED/wiki/Multiple-Controller-Examples

closed 2020-08-19 by QuitButton