FastLED/FastLED #1071: Add support for Adafruit Grand Central M4 (SAMD51)

spaceriqui, 2020-08-19:
Add support for Adafruit Grand Central M4 (SAMD51).  Currently using FastLED 3.3.3.

[Adafruit Grand Central M4 product page](https://www.adafruit.com/product/4064)

[Pinouts](https://learn.adafruit.com/adafruit-grand-central/pinouts)

`In file included from C:\../FastLED.h:41,`
`                 from C:\......ino:29:`
`C:\....\FastLED/led_sysdefs.h:53:2: error: #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."`
`   53 | #error "This platform isn't recognized by FastLED... yet.  See comments in FastLED/led_sysdefs.h for options."`
`      |  ^~~~~`
`exit status 1`
`Error compiling for board Adafruit Grand Central M4 (SAMD51).`

hkolbeck-streem, 2021-05-24:
If anyone has time to tackle this it would be much appreciated. I'm very new to the world of microcontrollers or I'd take a stab at it myself.

Also, there's a duplicate of this issue here: https://github.com/FastLED/FastLED/issues/920

spaceriqui, 2021-05-29:
You can use Circuit Python for the same thing instead of FastLed.  That's what I ended up doing.    https://learn.adafruit.com/adafruit-grand-central/what-is-circuitpython

bwmaas, 2023-09-09:
I'm willing to start working on this as I need support for this board...

Forking now for development. I'll try to report back in a day or two.

bwmaas, 2023-09-09:
Should have a fix in PR #1539 to get this platform working.